gopcua/opcua #21: Where to find ServiceType definitions?

zemirco, 2018-08-27:
Hey,

I'm currently trying to implement the `ReadRequest` service. I found your service type definitions in `services/service.go`.

https://github.com/wmnsk/gopcua/blob/23a591dbf62aa3f129b76dbd51b9883503bd1914/services/service.go#L13-L20

Where did you find the numbers? I couldn't find them in the spec. I found them here https://open62541.org/doc/open62541-0.3.pdf (page 170) and saw them in Wireshark. How did you know about them?

Best regards,
Mirco

wmnsk, 2018-08-27:
Hi,

I'm referring to this CSV, but I don't remember where in spec it was written... :disappointed: 
https://opcfoundation.org/UA/schemas/1.04/NodeIds.csv

Could you put the URL in comments when update `service.go` next?

zemirco, 2018-08-27:
ok, thank you!

closed 2018-08-27 by zemirco