gopcua/opcua #27: Method to retrieve Identifier value in NodeID

wmnsk, 2018-09-03:
When handling each type of NodeID, the main concern of the user is to retrieve values in `Identifier` field.
So, I'm considering to implement `Value()` or `Payload()` method on each NodeID, which makes the NodeID interface to return the value without asserting.

### Design in my mind

* Modify interface

```go
type NodeID interface {
	Serialize() ([]byte, error)
	SerializeTo([]byte) error
	DecodeFromBytes([]byte) error
	Len() int
	String() string
	EncodingMaskValue() uint8
	SetURIFlag()
	SetIndexFlag()
	Value() ([]byte, error) // Add this line
}
```

* Basic method

```go
func (n *NumericNodeID) Value() ([]byte, error) {
	if n == nil {
		return nil, errors.NewErrReceiverNil(n)
	}

	// return identifier as it is
	return n.Identifier, nil
}
```

* Special case for GUIDNodeID

```go
func (g *GUIDNodeID) Value() ([]byte, error) {
	if n == nil {
		return nil, errors.NewErrReceiverNil(g)
	}

	// return serialized GUID instead of GUID structure
	return n.GUID.Serialize()
}
```

wmnsk, 2018-09-03:
@zemirco 
I came up with this idea when looking into your PR #26.
Let me know if you have any thoughts on this :smiley:

zemirco, 2018-09-03:
Good idea! I had something similar in mind but didn't work it through. Problem is we have multiple `NodeID` types that all have different `Identifier` types.

- TwoByteNodeID -> `byte`
- FourByteNodeID -> `uint16`
- NumericNodeID -> `uint32`
- StringNodeID -> `[]byte`
- GUIDNodeID -> `*GUID`
- OpaqueNodeID -> `[]byte`

I would call the method `GetIdentifier()`. And I would drop the check for nil. I think the code will crash anyway before you can handle the error.

```go
func (o *OpaqueNodeID) GetIdentifier() []byte {
	return n.Identifier
}
```

wmnsk, 2018-09-03:
Thanks @zemirco.
I've worked on initial implementation of `GetIdentifier()` method in PR #29. I think it helps to some extent, but let me know if you have any additional idea to improve it.

zemirco, 2018-09-03:
Wow that was fast! Looks good to me :+1:

wmnsk, 2018-09-03:
Thanks! I've merged it.

commit a50e185f (2018-09-03): Implemented GetIdentifier() method in NodeIDs (#29), fixes #27

* impl GetIdentifier() method in NodeID

* removed nil check
closed 2018-09-03 by wmnsk