gopcua/opcua #59: GopherCon 2018: Matt Layher - Implementing a Network Protocol in Go

zemirco, 2018-09-14:
This is a great talk https://www.youtube.com/watch?v=pUaFW98V1Sc. The topic is very similar to what we're building. He even talks about go-fuzz.

One thing that came to mind. Our interface names are `Decode` and `Serialize`. He's using `Marshal` and `Unmarshal` like the json package https://golang.org/pkg/encoding/json/ and the proto package https://godoc.org/github.com/golang/protobuf/proto.

Therefore I'd like to propose changing our interface before releasing version 1.

wmnsk, 2018-09-14:
Nice proposal. I saw it before and it made me more eager to work on this project :smiley: 

Actually, the words `Decode` and `Serialize` are inspired by google/gopacket, and for me it was more familiar English word for this purpose.
However, I also think following the standard libraries and the leaders like @mdlayher is better, so let me see some examples around to consider more about it.

wmnsk, 2018-09-14:
FYI, I created GitHub project to collect items to be done before first release, not to miss something :100: 
https://github.com/wmnsk/gopcua/projects/2

mdlayher, 2018-09-14:
Glad you all enjoyed the talk!  Ultimately I think it's just important to keep your naming consistent where possible.  And definitely run go-fuzz on your code!  Best of luck on the project.

wmnsk, 2019-03-03:
Re-considered in the future, as the PR#124 gives a whole new paradigm.

closed 2019-03-03 by wmnsk