gopcua/opcua #79: Open the Wiki

zemirco, 2018-09-19:
Hey,

I tried adding some stuff to the wiki. Looks like it's not activated or I'm not allowed to use it. Could you change that?

Today I experimented with gopcua in the browser. I have a working example compiling it to webassembly and decoding raw messages in the browser.

The idea is to use gopcua in the browser and communicate with OPC UA servers that support websockets.

wmnsk, 2018-09-19:
I think it’s open to everyone according to the settings.
I opened a default wiki, so could you try it again?

[image]

zemirco, 2018-09-19:
I can see the wiki now but I cannot add content.

[image]

zemirco, 2018-09-19:
I'll try creating a new page locally and push it via git. wait a second.

wmnsk, 2018-09-19:
Ah I’ve overlooked the second checkbox in settings. 
Can you try on web again?

zemirco, 2018-09-19:
yes it works! thx

closed 2018-09-19 by zemirco