gopcua/opcua #81: error: unsupported uint16: unsupported or not implemented yet.

rowdyroad, 2018-10-01:
Hi!

I'm trying to connect from prosys opaua client (osx) to your example/server (with my changes of course).

Have an issue while connect?

2018/10/01 00:30:44 main.go:79: D Successfully established connection with 172.21.0.1:60570
2018/10/01 00:30:44 main.go:89: D Successfully opened secure channel with 172.21.0.1:60570
2018/10/01 00:30:44 main.go:95: E ReadService error: unsupported uint16: unsupported or not implemented yet.

Any Ideas?

Ready to collaborate.

wmnsk, 2018-10-01:
Hi, thank you for trying the sample and reporting the issue!

What OPC UA Service did you send from the client? `ReadService` reads Service above UASC and try to decode, and if the type of service is not implemented in our stack, it raises error.
For sure that non-user-friendly error message should be improved...

Here you have two options to address

1. Use `Read` instead of `ReadService`, which you can get Service in binary with UASC header.

2. Implement the service you want to use in `services` directory, and add it in `services.Decode()`, which requires some work depending on the complexity of the service.

rowdyroad, 2018-10-01:
Hi! Thanks for your answer!

I'm using 3d-party application - Prosys OPC UA Client.
So, I don't know what it sends, but right after i've pressed connect button I have this error (unsupported uint16: unsupported or not implemented yet - on server side of course.)

wmnsk, 2018-10-01:
I don’t know how it behaves either... Could you capture the packets and attach here?

rowdyroad, 2018-10-01:
Add some to  github.com/wmnsk/gopcua/services/service.go

```go
default:
		fmt.Fprintln(os.Stderr, "Unsupported id is", n.Identifier)
		fmt.Fprintf(os.Stderr, "Data: %X", b)
		return nil, errors.NewErrUnsupported(n.Identifier, "unsupported or not implemented yet.")
	}
```
**Received**

Unsupported id is 422
Data: 0100A601000000000000000000000000000000000000FFFFFFFFC0D40100000000190000006F70632E7463703A2F2F6C6F63616C686F73743A31343834300000000000000000

E ReadService error: unsupported uint16: unsupported or not implemented yet.

rowdyroad, 2018-10-01:
I've checked freeopcua.
It's a   FIND_ServerS_REQUEST = 0x1A6, // 422

wmnsk, 2018-11-04:
@rowdyroad 
Let me know if something's still go wrong. Thanks.

closed 2018-11-04 by wmnsk