gopcua/opcua #115: Code Generation

magiconair, 2018-10-15:
Hi @wmnsk, I had started working on a pure go opcua implementation in April but didn't make much headway until recently but you've beaten me to the punch :) 

My code is roughly in the same state (encoder, decoder, uacp, session, ...) but I think having a larger and active community is more important and your project has gained some good traction. I have some experience from that with my own open-source projects.

The one big thing I can contribute is a code generator for all OPC/UA service objects and array types and I'm also thinking of writing a reflection based Encoder/Decoder to reduce the boilerplate even further.

The generated code looks like the one below and also generates all length encoded Array variants of objects like `[]*ReadValueId`.

Would you be interested in this? If yes, then we can discuss what the interfaces should look like since the generated code isn't a drop-in replacement for the current structures. I'll attach a PR with the code generator so that you can have a look.

```go
package service

import (
	"bytes"

	ua "github.com/northvolt/go-opcua/datatypes"
)

type CreateSubscriptionRequest struct {
	RequestHeader               *RequestHeader
	RequestedPublishingInterval float64
	RequestedLifetimeCount      uint32
	RequestedMaxKeepAliveCount  uint32
	MaxNotificationsPerPublish  uint32
	PublishingEnabled           bool
	Priority                    uint8
}

func NewCreateSubscriptionRequest() *CreateSubscriptionRequest {
	return &CreateSubscriptionRequest{
		RequestHeader: NewRequestHeader(),
	}
}

func (m *CreateSubscriptionRequest) ID() uint16 {
	return 787
}

func (m *CreateSubscriptionRequest) GetName() string {
	return "CreateSubscriptionRequest"
}

func (m *CreateSubscriptionRequest) Read(b *bytes.Buffer) error {
	var err error
	if m.RequestHeader, err = ReadRequestHeader(b); err != nil {
		return err
	}
	if m.RequestedPublishingInterval, err = ua.ReadDouble(b); err != nil {
		return err
	}
	if m.RequestedLifetimeCount, err = ua.ReadUInt32(b); err != nil {
		return err
	}
	if m.RequestedMaxKeepAliveCount, err = ua.ReadUInt32(b); err != nil {
		return err
	}
	if m.MaxNotificationsPerPublish, err = ua.ReadUInt32(b); err != nil {
		return err
	}
	if m.PublishingEnabled, err = ua.ReadBoolean(b); err != nil {
		return err
	}
	if m.Priority, err = ua.ReadByte(b); err != nil {
		return err
	}
	return nil
}

func (m *CreateSubscriptionRequest) Write(b *bytes.Buffer) error {
	if err := WriteRequestHeader(b, m.RequestHeader); err != nil {
		return err
	}
	if err := ua.WriteDouble(b, m.RequestedPublishingInterval); err != nil {
		return err
	}
	if err := ua.WriteUInt32(b, m.RequestedLifetimeCount); err != nil {
		return err
	}
	if err := ua.WriteUInt32(b, m.RequestedMaxKeepAliveCount); err != nil {
		return err
	}
	if err := ua.WriteUInt32(b, m.MaxNotificationsPerPublish); err != nil {
		return err
	}
	if err := ua.WriteBoolean(b, m.PublishingEnabled); err != nil {
		return err
	}
	if err := ua.WriteByte(b, m.Priority); err != nil {
		return err
	}
	return nil
}

func ReadCreateSubscriptionRequest(b *bytes.Buffer) (*CreateSubscriptionRequest, error) {
	m := new(CreateSubscriptionRequest)
	if err := m.Read(b); err != nil {
		return nil, err
	}
	return m, nil
}

func WriteCreateSubscriptionRequest(b *bytes.Buffer, m *CreateSubscriptionRequest) error {
	return m.Write(b)
}
```

```go
package service

import (
	"bytes"

	ua "github.com/northvolt/go-opcua/datatypes"
)

type ReadValueIdArray []*ReadValueId

func (m ReadValueIdArray) GetName() string {
	return "ReadValueIdArray"
}

func ReadReadValueIdArray(b *bytes.Buffer) ([]*ReadValueId, error) {
	n, err := ua.ReadUInt32(b)
	if err != nil {
		return nil, err
	}
	if n == ua.NilValue {
		return nil, nil
	}
	m := make([]*ReadValueId, n)
	for i := uint32(0); i < n; i++ {
		if m[i], err = ReadReadValueId(b); err != nil {
			return nil, err
		}
	}
	return m, nil
}

func WriteReadValueIdArray(b *bytes.Buffer, m []*ReadValueId) error {
	if m == nil {
		return ua.WriteUInt32(b, ua.NilValue)
	}
	if err := ua.WriteUInt32(b, uint32(len(m))); err != nil {
		return err
	}
	for i := range m {
		if err := WriteReadValueId(b, m[i]); err != nil {
			return err
		}
	}
	return nil
}
```

magiconair, 2018-10-15:
Thank you. If this suits you then we can either adapt the code generator to generate the current style or we change the current code to the style of the code generator. 

I've spent some time wrangling with this and in the end arrived at the following guidelines:

* The generated code should use native go types where possible, i.e. `uint16` instead of `UInt16` to avoid casting in the code.
* The encode/decode code for structs should be boring and repetitive, i.e. no byte arithmetic and hence the use of `bytes.Buffer`. Using `io.Reader` and `io.Writer` methods would make testing a bit simpler but at the expense of managing allocations in a lot of places. I think your approach of documented byte arrays to test the codecs is better.
* The code should rely on as little hand-written code as possible, i.e. only the 27 basic datatypes are hand-coded and the rest (even the `[]StatusCode`) is generated.

This is why the structs have both `m.Read` and `m.Write` methods which do the heavy lifting and `ReadXXX()` and `WriteXXX()` functions which can be used in other structs. This is a bit similar to your `Serialize` and `SerializeTo` methods.

magiconair, 2018-10-16:
I've tried to generate code the way it is currently written and got quite far. I'll amend the PR when it is complete and we can discuss this further.

wmnsk, 2018-10-16:
OK, thanks!

magiconair, 2018-10-24:
I've been looking at the marshalling code for a while. This is what I see:

The current code crashes on decode if the buffer is too short since there are no length checks. 

We could add length checks but would have to do so for every field and before every operation which moves the offset. This adds a lot of boilerplate and is brittle if we continue to write the code by hand since there won't be enough tests to cover all cases.

I see three options:

1. Code generation 

    Generate marshalling code for all structs and arrays and include the length checks.

2. Use `bytes.Buffer`

    Instead of decoding from `[]byte` we can use `bytes.Buffer` which returns `io.ErrUnexpectedEOF` if you read past the buffer. This is better than crashing.

3. Use reflection

    Since we only have to traverse the type tree reflection should be the least amount of code and should work for almost all cases. I'd add a DecodeOPCUA() interface for complex cases like `Variant` and `DataValue` which have switch fields but none of the service structs has them. 

The other thing are the OPC/UA specific types like `Float` and `UInt32` as well as custom classes for all the arrays like `StringArray`. Since Go slices already have a length this feels like double work. 

I was able to implement a reflection based codec within a day in about 600 lines of code. This should be able to replace most of the 12k LOC in `datatypes` and `services`. I'll push the code as a separate PR and try to integrate it with the current data structures.

sbinet, 2018-10-25:
alternatively, one could have a reader/decoder that caches an error (if any) and returns the zero value in case of an error.

that's usually what I do for this kind of code:
- https://github.com/go-boostio/boostio/blob/master/binser/rbuffer.go#L108

magiconair, 2018-10-25:
That's what I initially started out with. However, you'd probably still want to generate the code from the XML file - especially for the slice types - and then it doesn't really matter. 

Also, if you use a buffer with methods for the builtin types the decoding functions become a mix of method and function calls, e.g.:

```go
type A struct {
    N uint32
    B *B
}

func (a *A) Decode(b *Buffer) error {
    a.N = b.ReadUint32()
    a.B = ReadB(b) // do you alloc B here or in Decode?
    return b.Err()
}
```

If you use helper functions for every type, then you can add more helpers as you need and the style does not change.

```go
type A struct {
    N uint32
    B *B
}

func (a *A) Decode(r io.Reader) error {
    var err error
    if a.N, err = ReadUint32(r); err != nil {
        return err
    }
    if a.B, err = ReadB(r); err != nil {
        return err
    }
    return nil
}
```

That is simpler to write and generate, IMO.

magiconair, 2018-10-25:
The reflection based codec has `DecodeOPCUA` and `EncodeOPCUA` methods as an escape hatch for types with switches like `NodeID`. Requiring them for the builtins means that you either need to shadow all builtin types (e.g. `UInt32`) just for decoding or do the cast to an `Uint32` during decode. With the reflection codec you can actually remove most of the code for the builtins.

Also, it seems nicer to use `[]*B` instead of `type BArray []*B` since all arrays are length encoded anyway.

I'm currently integrating the reflection codec with the existing code to show that this works. This requires a bit of refactoring of the `NodeID` to make this a struct instead of a polymorphic structure. I'll submit this as a separate PR. 

I'm more than happy to continue the conversation but would prefer if we base this on code that is more functional. Would this be OK?

sbinet, 2018-10-25:
sure :)

magiconair, 2019-02-26:
I'm going to close this one since the reflection codec in #124 seems to be working and it is possible to either hand-write or generate custom encoder/decoder functions.

PR #124 (2019-03-13): reflection codec
This patch provides a reflection based implementation of the OPC/UA
binary protocol encoding scheme.

Fixes #115
closed 2019-02-26 by magiconair