gopcua/opcua #129: ERROR on	go secChan.monitor(ctx)

abay94, 2018-10-30:
Please could you help me on this issue

[image]

wmnsk, 2018-10-30:
This might be due to the unexpected format of a message coming.
Could you share your source code in `main.go` and pcap file if available?

abay94, 2018-10-30:
Thank you for your response. 
I have just run the gopcua/examples/client/client.go script with changing endpoint.

p.s. I am  a new in golang ( sorry if it is so stupid question)

`// Copyright 2018 gopcua authors. All rights reserved.
// Use of this source code is governed by a MIT-style license that can be
// found in the LICENSE file.

/*
Command client provides a connection establishment of OPC UA Secure Conversation.
*/
package main

import (
	"context"
	"encoding/hex"
	"flag"
	"log"
	"time"

	"github.com/wmnsk/gopcua/datatypes"
	"github.com/wmnsk/gopcua/services"
	"github.com/wmnsk/gopcua/uacp"
	"github.com/wmnsk/gopcua/uasc"
	"github.com/wmnsk/gopcua/utils"
)

func main() {
	var (
		endpoint   = flag.String("endpoint", "opc.tcp://Abays-MacBook-Pro.local:53530/OPCUA/SimulationServer", "OPC UA Endpoint URL")
		payloadHex = flag.String("payload", "ff13", "Payload to send in hex stream format")
	)
	flag.Parse()

	// Create context for UACP to be used by statemachine working background.
	ctx := context.Background()
	ctx, cancel := context.WithCancel(ctx)
	defer cancel()

	// Establish UACP Connection with the Endpoint specified.
	conn, err := uacp.Dial(ctx, *endpoint)
	if err != nil {
		log.Fatal(err)
	}
	defer func() {
		if err := conn.Close(); err != nil {
			log.Fatalf("Failed to shutdown connection: %s", err)
		}
		log.Println("Successfully shutdown connection")
	}()
	log.Printf("Successfully established connection with %v", conn.RemoteEndpoint())

	// Open SecureChannel on top of UACP Connection established above.
	cfg := uasc.NewClientConfigSecurityNone(3333, 3600000)
	secChan, err := uasc.OpenSecureChannel(ctx, conn, cfg, 5*time.Second, 3)
	if err != nil {
		log.Fatal(err)
	}
	defer func() {
		if err := secChan.Close(); err != nil {
			log.Fatalf("Failed to close secure channel: %s", err)
		}
		log.Printf("Successfully closed secure channel with %v", conn.RemoteEndpoint())
	}()
	log.Printf("Successfully opened secure channel with %v", secChan.RemoteEndpoint())

	//Send FindServersRequest to remote Endpoint.
	if err := secChan.FindServersRequest([]string{"ja-JP", "de-DE", "en-US"}, "gopcua-server"); err != nil {
		log.Fatal(err)
	}
	log.Println("Successfully sent FindServersRequest")
	time.Sleep(500 * time.Millisecond)

	// Send GetEndpointsRequest to remote Endpoint.
	if err := secChan.GetEndpointsRequest([]string{"ja-JP", "de-DE", "en-US"}, []string{"gopcua-server"}); err != nil {
		log.Fatal(err)
	}
	log.Println("Successfully sent GetEndpointsRequest")
	time.Sleep(500 * time.Millisecond)

	sessCfg := uasc.NewClientSessionConfig(
		[]string{"en-US"},
		datatypes.NewAnonymousIdentityToken("anonymous"),
	)
	session, err := uasc.CreateSession(ctx, secChan, sessCfg, 3, 5*time.Second)  ////// <<<<<<<<  errrrorrrr
	if err != nil {
		log.Fatal(err)
	}
	defer func() {
		session.Close()
		log.Printf("Successfully closed secure channel with %v", secChan.RemoteEndpoint())
	}()
	log.Printf("Successfully created secure channel with %v", secChan.RemoteEndpoint())

	if err := session.Activate(); err != nil {
		log.Fatal(err)
	}
	log.Printf("Successfully activated secure channel with %v", secChan.RemoteEndpoint())

	if err := session.ReadRequest(
		2000, services.TimestampsToReturnBoth, datatypes.NewReadValueID(
			datatypes.NewNumericNodeID(0, 11111), datatypes.IntegerIDValue, "", 0, "",
		),
	); err != nil {
		log.Fatal(err)
	}
	log.Println("Successfully sent ReadRequest")

	// Send arbitrary payload on top of UASC SecureChannel.
	payload, err := hex.DecodeString(*payloadHex)
	if err != nil {
		log.Fatal(err)
	}
	if _, err := secChan.WriteService(payload); err != nil {
		log.Fatal(err)
	}
	log.Printf("Successfully sent message: %x\n%s", payload, utils.Wireshark(0, payload))
	time.Sleep(500 * time.Millisecond)
}`

lishuaia, 2018-10-31:
hi abay94
 I have been looking at the code of the block recently. Can I leave a way of communication?

wmnsk, 2018-10-31:
OK it seems the source code is the same one in `examples` directory.
Then we need to analyze the incoming packets to find out the exact cause. Could you reproduce it in your environment again and share the captured packets?

abay94, 2018-10-31:
Helllooo! 
Here is my print 
[image]

[image]

abay94, 2018-10-31:
Yes, of course thanks!

wmnsk, 2018-11-01:
@abay94 
Could you share the packet captures, not the source codes and console outputs?

abay94, 2018-11-01:
Sorry, I do not know where can I see the captured packets. If the next msg is not a packet ( it is decoded part of attribute of SucureChannel:rcvBuf, 3 times) 

`**1)** &{MessageType: 5197902, ChunkType: 70, MessageSize: 135, Payload: 2f000000687474703a2f2f6f7063666f756e646174696f6e2e6f72672f55412f5365637572697479506f6c696379234e6f6e65ffffffffffffffffe8030000050d00000100c1010000000000000000010000000000000000ffffffff00000000000000240000000100000000e4fd2b7970d40180ee360000000000, SecurityPolicyURI: 47, http://opcfoundation.org/UA/SecurityPolicy#None, SenderCertificate: &{-1 []}, ReceiverCertificateThumbprint: &{-1 []}, Payload: e8030000050d00000100c1010000000000000000010000000000000000ffffffff00000000000000240000000100000000e4fd2b7970d40180ee360000000000 <nil> SequenceNumber: 1000, RequestID: 3333, Payload: 0100c1010000000000000000010000000000000000ffffffff00000000000000240000000100000000e4fd2b7970d40180ee360000000000 2185-07-21 23:34:33.709551616 +0000 UTC, 1, 0, [0], &{-1 []}, &{0, 0 0}, 00000000240000000100000000e4fd2b7970d40180ee360000000000}

**2)** &{MessageType: 5067591, ChunkType: 70, MessageSize: 56, Payload: 01000000e9030000050d00000100a90160cefe2b7970d4010200000000000000000000000000000000000000, <nil> TokenID: 1, Payload: e9030000050d00000100a90160cefe2b7970d4010200000000000000000000000000000000000000 SequenceNumber: 1001, RequestID: 3333, Payload: 0100a90160cefe2b7970d4010200000000000000000000000000000000000000 2018-10-30 17:51:23.206 +0000 UTC, 2, 0, [0], &{0 []}, &{0, 0 0}, 00000000}

**3)** &{MessageType: 5067591, ChunkType: 70, MessageSize: 56, Payload: 01000000ea030000050d00000100af0120524c2c7970d401030000000000000000ffffffff00000000000000, <nil> TokenID: 1, Payload: ea030000050d00000100af0120524c2c7970d401030000000000000000ffffffff00000000000000 SequenceNumber: 1002, RequestID: 3333, Payload: 0100af0120524c2c7970d401030000000000000000ffffffff00000000000000 2018-10-30 17:51:23.714 +0000 UTC, 3, 0, [0], &{-1 []}, &{0, 0 0}, 00000000} `

lishuaia, 2018-11-01:
hi @abay94 
Are you using the go language to write the opcua client, and then read the data from the server? Can you give me your mail box? Let's talk about it.

wmnsk, 2018-11-01:
From the logs you pasted, gopcua tried to decode `GetEndpointsResponse` with "Good(0x00000000)" `ServiceResult` but it has zero-sized `Endpoints`. So this might be the problem with our decoder causing crash when decoding it(Decoders are not so robust in the current implementation).

* If you don't need to send `GetEndpointsRequest` to your server, delete the lines related to it.
* If you need it, please capture packets and attach here to debug. Packets can be captured with Wireshark or tcpdump.

abay94, 2018-11-01:
OK thanks!!

closed 2018-11-03 by wmnsk