gopcua/opcua #130: Opcua Client reads server data

lishuaia, 2018-10-31:
Hi I've been looking at your code recently, but I'm not sure how you read server data with client, and I haven't found the location to set the identifier, identifier type, and Namespace Index parameters, can you tell me?

wmnsk, 2018-10-31:
Hi @lishuaia 

In short, there's no feature to realize that in this library.
Currently we're working on designing and implementing OPC UA protocol stack itself, and such a high-level feature is not available.

There are two ways to meet your demands, both of which require some efforts, though.
* Implement APIs into package itself.
Implementing some helper APIs helps users a lot if you're interested in contributing our project.

* Use gopcua just as a protocol stack.
Even with the current implementation, you can write your own program to write/hold/return some values via API, using the values inside each packets.

lishuaia, 2018-10-31:
hi @wmnsk  
I wrote the code of opcua in Java language.No opcua code written in GoLand.Opcua also has no go SDK.
Should the client read data without setting up a Identifer? The java code implements that the client reads the server's data, specifying the identifier, identifier type, and Namespace to read the data. What API can go with?

wmnsk, 2018-10-31:
I think there's no API to achieve it currently. This project is a library to provide protocol stack features for higher-level application developers, not the independent application to let users handle everything related to OPC UA.

To avoid misunderstanding, can you show me some examples in another API implementation(in other languages) that meets your demands? Then we can consider implementing it or not.

lishuaia, 2018-10-31:
Here is an example.
https://github.com/streamsets/datacollector/tree/master/basic-lib/src/main/java/com/streamsets/pipeline/stage/origin/opcua

wmnsk, 2018-10-31:
Can you specify the exact APIs that application developers use?
I don't really understand what you're trying to do.

lishuaia, 2018-10-31:
I want to use go language to write a opcua client to read server data.

wmnsk, 2018-10-31:
OK, then what you need to do is;

* Copy `examples/client/client.go`.
* Delete the lines which is not concerned to `ReadRequest`
* Modify parameters in `session.ReadRequest()` as your server likes
* Retrieve values in `ReadResponse` from server with `session.Read()` and type assertion

I think `datatypes.ParseNodeID()` function also helps you construct your ReadRequest parameters.

And, please note currently the `Variant` supports only a few number of types to decode. If the type of value returned from server is unsupported one, it won't work for you.
https://github.com/wmnsk/gopcua/blob/master/datatypes/variant.go#L42

lishuaia, 2018-11-01:
After I set the parameters of session.ReadRequest (), how can I read the data from the server? Here is my code to install your example. Can you help me see what's wrong with it?

/*
Command client provides a connection establishment of OPC UA Secure Conversation.
*/
package main

import (
	"context"
	"flag"
	"fmt"
	"log"
	"time"

	"github.com/wmnsk/gopcua/datatypes"
	"github.com/wmnsk/gopcua/services"
	"github.com/wmnsk/gopcua/uacp"
	"github.com/wmnsk/gopcua/uasc"

)

func main() {
	var (
		//object/Demo/Dynamic/Arrays
		endpoint   = flag.String("endpoint", "opc.tcp://desktop-buubend:48030/", "OPC UA Endpoint URL")
	)
	flag.Parse()

	// Create context for UACP to be used by statemachine working background.
	ctx := context.Background()
	ctx, cancel := context.WithCancel(ctx)
	defer cancel()

	// Establish UACP Connection with the Endpoint specified.
	conn, err := uacp.Dial(ctx, *endpoint)
	if err != nil {
		log.Fatal(err)
	}
	defer func() {
		if err := conn.Close(); err != nil {
			fmt.Println("第二个defer")
			log.Fatalf("Failed to shutdown connection: %s", err)
		}
		log.Println("Successfully shutdown connection")
	}()
	log.Printf("Successfully established connection with %v", conn.RemoteEndpoint())

	// Open SecureChannel on top of UACP Connection established above.
	cfg := uasc.NewClientConfigSecurityNone(3333, 3600000)
	secChan, err := uasc.OpenSecureChannel(ctx, conn, cfg, 5*time.Second, 3)
	if err != nil {
		log.Fatal(err)
	}
	defer func() {
		if err := secChan.Close(); err != nil {

			log.Fatalf("Failed to close secure channel: %s", err)
		}
		log.Printf("secChan Successfully closed secure channel with %v", conn.RemoteEndpoint())
	}()
	log.Printf("Successfully opened secure channel with %v", secChan.RemoteEndpoint())

	// Send FindServersRequest to remote Endpoint.
	if err := secChan.FindServersRequest([]string{"ja-JP", "de-DE", "en-US"}, "gopcua-server"); err != nil {
		log.Fatal(err)
	}
	log.Println("Successfully sent FindServersRequest")
	time.Sleep(500 * time.Millisecond)

	// Send GetEndpointsRequest to remote Endpoint.
	if err := secChan.GetEndpointsRequest([]string{"ja-JP", "de-DE", "en-US"}, []string{"gopcua-server"}); err != nil {
		log.Fatal(err)

	}
	log.Println("Successfully sent GetEndpointsRequest")
	time.Sleep(500 * time.Millisecond)

	sessCfg := uasc.NewClientSessionConfig(
		[]string{"ja-JP"},
		datatypes.NewAnonymousIdentityToken("anonymous"),
	)
	session, err := uasc.CreateSession(ctx, secChan, sessCfg, 3, 5*time.Second)
	if err != nil {
		log.Fatal(err)
	}
	defer func() {
		session.Close()
		log.Printf("session Successfully closed secure channel with %v", secChan.RemoteEndpoint())
	}()
	log.Printf("Successfully created secure channel with %v", secChan.RemoteEndpoint())

	if err := session.Activate(); err != nil {
		log.Fatal(err)
	}
	log.Printf("Successfully activated secure channel with %v", secChan.RemoteEndpoint())

	id := datatypes.NewStringNodeID(2, "Demo.Dynamic.Scalar.UInt64")
	if err := session.ReadRequest(
		2000,services.TimestampsToReturnBoth,datatypes.NewReadValueID(
			id,datatypes.IntegerIDValue,"",0,"",
			),
		);err !=nil{
		log.Fatal(err)
	}
	log.Println("Successfully sent ReadRequest")
	n, e := session.Read([]byte(id.Value()))
	if e!=nil {
		fmt.Println("session.Read",e)
	}
	fmt.Println("n::::",n)
	
	
}

lishuaia, 2018-11-01:
sorry，I don't understand. Can you help me from the code I wrote?
Now I'm going to set up session.ReadRequest, how do I use session.ReadResponse and session.Read ()? Can you write an example for me? Thank you

Code：

id := datatypes.NewStringNodeID(2, "Demo.Dynamic.Scalar.UInt64")
if err := session.ReadRequest(
	2000,services.TimestampsToReturnBoth,datatypes.NewReadValueID(
		id,datatypes.IntegerIDValue,"",0,"",
		),
	);err !=nil{
	log.Fatal(err)
}
log.Println("Successfully sent ReadRequest")
n, e := session.Read([]byte(id.Value()))
if e!=nil {
	fmt.Println("session.Read",e)
}
fmt.Println("n::::",n)

wmnsk, 2018-11-01:
Sorry but that is no longer the question about our library. Please read the document I posted above carefully and find what parameters and returned values in `Read()` means.
[Stack Overflow](https://stackoverflow.com/) might be better place to ask.

Hope you can succeed developing your client application 😃

closed 2018-11-01 by wmnsk

abay94, 2018-11-19:
> OK, then what you need to do is;
>
> * Copy `examples/client/client.go`.
> * Delete the lines which is not concerned to `ReadRequest`
> * Modify parameters in `session.ReadRequest()` as your server likes
> * [...]

Hello, 
Please could you help me with clarifying  "Retrieve values in `ReadResponse` from server"