gopcua/opcua #135: ReadResponse service

abay94, 2018-11-20:
Hello,
question regards to https://github.com/wmnsk/gopcua/issues/130#issuecomment-434630675

Please could you help me with clarifying "Retrieve values in ReadResponse from server"
Did you mean, getting data from services.DecodeReadResponse(buf) and then getting from the struct of ReadResponse.Result?
I did session.Read() and variant EncodingMask  shows the right type from server. But other parameters are &{10 <nil> <nil> <nil> []}

wmnsk, 2018-12-03:
Could you share your code?

abay94, 2018-12-03:
I think i did retrieve data, thank you. Is it true?
```
ourNodeID, err := datatypes.ParseNodeID(tagName)
		if err != nil {
			fmt.Printf("There is a problem on parseNodeId : %s", err)
		}

		if err := session.ReadRequest(
			2000, services.TimestampsToReturnBoth, datatypes.NewReadValueID(
				ourNodeID, datatypes.IntegerIDValue, "", 0, ""),
		); err != nil {
			log.Fatal(err)
		}

		data := make([]byte, 1000)

		_, e := session.Read(data)
		if e != nil {
			fmt.Println("session.Read", e)
		}
typeOfTag := data[57]
	fields := make(map[string]interface{})
	if typeOfTag == 10 {
		bits := binary.LittleEndian.Uint32(data[58:62])
		value := math.Float32frombits(bits)
		fields[tag_name] = value
		//fmt.Printf("There is a value of %d -->>   %v \n", typeOfTag, value)
	} else if typeOfTag == 11 {
		bits := binary.LittleEndian.Uint64(data[58:66])
		value := math.Float64frombits(bits)
		fields[tag_name] = value
		//fmt.Printf("There is a value of %d -->>   %v \n", typeOfTag, value)
	} else if typeOfTag == 6 {
		value := data[58]
		fields[tag_name] = value
		//fmt.Printf("There is a value of %d -->>  %v \n", typeOfTag, value)
	} else {
		fmt.Println("There is unknown type of tag!!!")
	}
	if len(fields) == 0 {
		fields[tag_name] = 0
	}
```

wmnsk, 2018-12-05:
It’s due to the error in decoding CreateSessionResponse message.
Can you share the pcap file so that I can see the cause more precisely?

abay94, 2018-12-19:
> It’s due to the error in decoding CreateSessionResponse message.
> Can you share the pcap file so that I can see the cause more precisely?

Thank you for your response! 

I did catch all packets and attached it (in zip file)
[All_packets.pcapng.zip](https://github.com/wmnsk/gopcua/files/2694262/All_packets.pcapng.zip)

[image]

wmnsk, 2018-12-21:
I worked on reproducing the crash but it didn’t.
Is the pcap and the log collected at the exactly same time?

abay94, 2018-12-21:
Yes it is collected at the the same time

marcotuna, 2018-12-22:
Hello, the same thing is happening with me. I tried using a public opcua server.
`go run client.go --endpoint "opc.tcp://opcuaserver.com:4840"`

```
2018/12/22 14:42:02 Successfully established connection with opc.tcp://opcuaserver.com:4840
2018/12/22 14:42:03 Successfully opened secure channel with opc.tcp://opcuaserver.com:4840
2018/12/22 14:42:03 Successfully sent FindServersRequest
2018/12/22 14:42:03 Successfully sent GetEndpointsRequest
panic: runtime error: slice bounds out of range

goroutine 20 [running]:
github.com/wmnsk/gopcua/services.(*EndpointDescription).DecodeFromBytes(0xc0001001c0, 0xc00015c038, 0x570, 0xffc7, 0xc000108480, 0x0)
	/home/unknown/go/src/github.com/wmnsk/gopcua/services/endpoint-description.go:82 +0x633
github.com/wmnsk/gopcua/services.DecodeEndpointDescription(0xc00015c038, 0x570, 0xffc7, 0x4f615e, 0xc000108460, 0xc000040d98)
	/home/unknown/go/src/github.com/wmnsk/gopcua/services/endpoint-description.go:46 +0x62
github.com/wmnsk/gopcua/services.(*EndpointDescriptionArray).DecodeFromBytes(0xc000108480, 0xc00015c034, 0x574, 0xffcb, 0x0, 0x0)
	/home/unknown/go/src/github.com/wmnsk/gopcua/services/endpoint-description.go:244 +0xed
github.com/wmnsk/gopcua/services.(*GetEndpointsResponse).DecodeFromBytes(0xc000108400, 0xc00015c018, 0x590, 0xffe7, 0x0, 0x0)
	/home/unknown/go/src/github.com/wmnsk/gopcua/services/get-endpoints-response.go:58 +0x277
github.com/wmnsk/gopcua/services.Decode(0xc00015c018, 0x590, 0xffe7, 0xffef, 0x0, 0x0, 0xc0000f6630)
	/home/unknown/go/src/github.com/wmnsk/gopcua/services/service.go:113 +0x1bd
github.com/wmnsk/gopcua/uasc.(*Message).decodeMSGFromBytes(0xc0000f6600, 0xc00015c00c, 0x59c, 0xfff3, 0xc000040ee8, 0x3)
	/home/unknown/go/src/github.com/wmnsk/gopcua/uasc/message.go:150 +0x116
github.com/wmnsk/gopcua/uasc.(*Message).DecodeFromBytes(0xc0000f6600, 0xc00015c000, 0x5a8, 0xffff, 0x5a8, 0x5a8)
	/home/unknown/go/src/github.com/wmnsk/gopcua/uasc/message.go:112 +0x2e0
github.com/wmnsk/gopcua/uasc.Decode(0xc00015c000, 0x5a8, 0xffff, 0xffff, 0x5a8, 0x0)
	/home/unknown/go/src/github.com/wmnsk/gopcua/uasc/message.go:90 +0x62
github.com/wmnsk/gopcua/uasc.(*SecureChannel).monitor(0xc00010c200, 0x5ab680, 0xc0000221c0)
	/home/unknown/go/src/github.com/wmnsk/gopcua/uasc/secure-channel.go:241 +0x110
created by github.com/wmnsk/gopcua/uasc.OpenSecureChannel
	/home/unknown/go/src/github.com/wmnsk/gopcua/uasc/client.go:53 +0x4d7
exit status 2
```

wmnsk, 2018-12-23:
OK thank youfor reporting @abay94 and @marcotuna 
Can you tell me the platform you’re using?

marcotuna, 2018-12-23:
@wmnsk I am using Linux Mint 19.1 which is based on Ubuntu 18.04 LTS

abay94, 2018-12-23:
Thank you!
Mac os high sierra v 10.13.5

magiconair, 2019-03-23:
@abay94 and @marcotuna please have a look at the code in #152 This will work when #153 is merged tomorrow or so.

magiconair, 2019-04-09:
Code is merged. Could you update your code and re-open this issue if necessary?

closed 2019-04-09 by magiconair