gopcua/opcua #136: Development timeline

felix-lessoer, 2019-02-04:
Hi gopcua team,

I really like your implementation in this package.
I would also like to use the browse services and I also need support for some more datatypes of the variant. 
I regularly run into this error while decoding the ReadResponse message: cannot decode as uint8: got undefined type

Can you share your timeline? Are there any plans to complete the list of services and available datatypes?

Thanks
Felix

magiconair, 2019-02-08:
I have a full set of datatypes ready for the Variant but the change isn't ready to be merged yet, unfortunately. However, until then you should be able to add the missing datatype to 

https://github.com/wmnsk/gopcua/blob/master/datatypes/variant.go#L46-L51

You probably need to add another basic `uint8` datatype as well. Similar to https://github.com/wmnsk/gopcua/blob/master/datatypes/uint32.go

felix-lessoer, 2019-02-08:
Well, I noticed that the error message is wrong. It refers the the datatype of v.EncodingMask which is uint8. After debugging into the code and see what Encoding Mask actually comes back it turned out that int32 and double was missing in my case.
I implemented these two types on my own to finish the evaluation. It is working fine for me now. But would be great to get your full list of datatypes instead of developing each one on my own. 

Do you still working on this project? Based on the timeline it looks like development has stopped last year.

magiconair, 2019-02-11:
I am still working on this but have other things to focus on. I am very keen on having a native OPC/UA implementation in Go.

wmnsk, 2019-02-11:
@felix-lessoer 
Thank you for trying our implementation!
Unfortunately I cannot show you the nicely aligned timeline but for sure it will go forward to be more robust/sufficient in the coming months, as from mid Februrary I am going to put some of my resources again.

wmnsk, 2019-02-11:
@magiconair
Reflection-based implementation in your PR would be the key to accelerate. If you don't have time right now, let's work on that together. I'll put some thoughts on #124.

magiconair, 2019-02-11:
yes, commented on #124 Lets push this together. I could really use this library right now.

magiconair, 2019-03-20:
@felix-lessoer the new codec has been merged and all types should now be supported. `Variant` may still have some edge cases not covered. Please note that there is now a high-level client but I need to properly bubble up the errors. This should probably land somewhere next week. 

It'd be cool if you can test the new code and provide some feedback. I'll close this one in the meantime. Feel free to re-open if necessary.

closed 2019-03-20 by magiconair