gopcua/opcua #142: Clarify supported features

strowk, 2019-03-07:
I cannot understand - is x509 Authentication supported or not? 
It seems to presented in code 
https://github.com/wmnsk/gopcua/blob/72c87ce6fabe6993089b948be572e56fe94670c5/datatypes/user-identity-token.go#L253
and even tested, but supported features do not mention it as supported. 
So I'm confused - was it supported not fully or its just not working code or it just a documentation issue?

If it's the last one I think other features should be checked too (maybe more of them supported already).
Thanks!

wmnsk, 2019-03-07:
Hi, thank you for having some interest in our project😀

The code you pointed out is just to decode the raw bytes into `X509IdentityToken` struct, which does not mean the certificate can be handled properly in the exchanges of packets.

So, with some(or even so much) effort you can use it, but as a library, we should tell you it has not been supported as an official feature yet, as the proper handling of certificates cannot be done only with deserializing what the endpoint received.

strowk, 2019-03-07:
Thanks for answer! Now I see that documentation is correct and I just had a wishful thinking on my side =)

closed 2019-03-07 by strowk

wmnsk, 2019-03-07:
Hope you’ll come back when it’s ready. Thanks.