gopcua/opcua #143: Default installation path conflicts with default expectation

strowk, 2019-03-07:
By default from sources all would be installed
https://opcuastack.readthedocs.io/en/latest/getting_started/installation.html
to 
`~/.ASNeG`
, which would require from user to do something like 
```
export PATH=$HOME/.ASNeG/usr/bin:$PATH
export LD_LIBRARY_PATH=$HOME/.ASNeG/usr/lib:$LD_LIBRARY_PATH
```
and it still would not really help, because when doing
https://opcuastack.readthedocs.io/en/latest/getting_started/hello_world.html
user would get the
```
CMake Error at CMakeLists.txt:82 (include):
  include could not find load file:

    /usr/share/cmake/OpcUaStack3/CMakeOpenSSL.txt
```

So either detailed explanation how to not get these problems should be included in documentation (what to add to .bashrc, how to configure currently opened shell, etc) or default path should be `/` instead `~/.ASNeG`. It seems that application is designed to be installed to root...

wmnsk, 2019-03-07:
It does not seem the documents and what you did is related to our project :confused:

closed 2019-03-07 by wmnsk

strowk, 2019-03-07:
Oh! Big sorry! Not this github =)