gopcua/opcua #152: Value nil with ReadResponse service in new version

felix-lessoer, 2019-03-22:
Hi,

I tried to implement the new version you released in my current project here
https://github.com/felix-lessoer/Machinebeat/blob/v2/beater/client.go

The problem is:
After changing the implementation to use your new client implementation I got no value from ReadResponse anymore.
Trying to get ReadResponse.Results[].Value always leads into nil.

I'm using this public server to test:
endpoint: "opc.tcp://opcuaserver.com:48010"

Node:
id: "AirConditioner_1.Temperature"
namespace: 3

How do you expect to build the ReadRequest?

`
        node = client.Node(nodeID)

	rh := ua.RequestHeader{}

	rv := ua.ReadValueID{}

	rv.NodeID = nodeID
	rv.AttributeID = 0
	rv.IndexRange = ""

	qn, err := node.BrowseName()
	if err != nil {
		return nil, err
	}
	rv.DataEncoding = qn

	rr := ua.ReadRequest{}
	rr.MaxAge = 2000
	rr.NodesToRead = []*ua.ReadValueID{&rv}
	rr.TimestampsToReturn = ua.TimestampsToReturnBoth
	rr.RequestHeader = &rh

	m, err := client.Read(&rr)
	if err != nil {
		return nil, err
	}
	value, status := handleReadResponse(m)
`

wmnsk, 2019-03-22:
Hi! Thanks trying our implementation!

Just to make sure, what that public server is expected to return?
If PCAP is provided it’s very useful and can accelerate the analysis.

magiconair, 2019-03-22:
Hi @felix-lessoer, 

I haven't tested `Read` fully yet and it is not handling the status code of the results yet. Generally speaking the code should look similar to https://github.com/gopcua/opcua/blob/master/node.go#L51-L63.

After adapting your code should look like this:

```go
req := &ua.ReadRequest{
    MaxAge: 2000,
    NodesToRead: []*ua.ReadValueID{
        &ua.ReadValueID{
            NodeID : nodeID,
            DataEncoding: &ua.QualifiedName{},
        },
    },
    TimestampsToReturn: ua.TimestampsToReturnBoth,
}

resp, err := client.Read(req)
if err != nil {
	return nil, err
}
if resp.Results[0].Status != ua.StatusOK {
    return nil, resp.Results[0].Status
}
value := resp.Results[0].Value
...

```

magiconair, 2019-03-22:
Note that you do not have to set the `RequestHeader`.

magiconair, 2019-03-23:
## Update

Forgot to set the `AttributeID` to tell the server to return the value. I've updated the code above. Here is a fully working example:

```go
// Copyright 2018-2019 opcua authors. All rights reserved.
// Use of this source code is governed by a MIT-style license that can be
// found in the LICENSE file.

package main

import (
	"flag"
	"log"

	"github.com/gopcua/opcua"
	"github.com/gopcua/opcua/ua"
)

func main() {
	endpoint := flag.String("endpoint", "opc.tcp://localhost:4840", "OPC UA Endpoint URL")
	nodeID := flag.String("node", "", "NodeID to read")
	flag.Parse()

	c := opcua.NewClient(*endpoint, nil)
	if err := c.Open(); err != nil {
		log.Fatal(err)
	}
	defer c.Close()

	id, err := ua.NewNodeID(*nodeID)
	if err != nil {
		log.Fatalf("invalid node id: %v", err)
	}

	req := &ua.ReadRequest{
		MaxAge: 2000,
		NodesToRead: []*ua.ReadValueID{
			&ua.ReadValueID{
				NodeID:       id,
				AttributeID:  ua.IntegerIDValue,
				DataEncoding: &ua.QualifiedName{},
			},
		},
		TimestampsToReturn: ua.TimestampsToReturnBoth,
	}

	resp, err := c.Read(req)
	if err != nil {
		log.Fatalf("Read failed: %s", err)
	}
	if resp.Results[0].Status != ua.StatusOK {
		log.Fatalf("Status not OK: %v", resp.Results[0].Status)
	}
	log.Printf("Value: %v", resp.Results[0].Value)
}
```

magiconair, 2019-03-23:
I'll add that example later today.

magiconair, 2019-03-23:
I'll probably update `Read` to set the `AttributeID` to return the value unless some other value is set and to also set an empty value for `DataEncoding` if not set to make the code simpler.

magiconair, 2019-04-09:
@felix-lessoer is this still an issue? I'll close this issue but feel free to comment/re-open if this is still a problem.

closed 2019-04-09 by magiconair