gopcua/opcua #163: encoder should be smart about empty structs

magiconair, 2019-04-08:
The OPC/UA binary protocol has no mechanism to express a `nil` struct. This means that all struct fields at least represent a Go zero value for that struct. Additionally, the current code generator creates struct fields as pointers which has the effect that all struct fields need to be initialized with at least an empty value. 

Here is an example from the `ua/write_response_test.go`.

```go
			Struct: &WriteResponse{
				ResponseHeader: &ResponseHeader{
					Timestamp:     time.Date(2018, time.August, 10, 23, 0, 0, 0, time.UTC),
					RequestHandle: 1,
					ServiceDiagnostics: &DiagnosticInfo{},
					StringTable:        []string{},
					AdditionalHeader:   NewExtensionObject(nil),
				},
				Results: []StatusCode{StatusOK},
			},
```

This behavior was also the cause for a discrepancy in #162 where a field of the session configuration had not been initialized. 

One fix is to make the encoder smart about `nil` values. That has the effect that tests can no longer use the same structs for encoding and decoding since a `nil` silently turns into an empty struct. 

A better approach might be not to generate struct fields as pointers since they cannot be `nil` anyway. Then the codec just works.

magiconair, 2019-04-08:
Switching the codec from pointer to value semantics is another all-or-nothing massive change but this feels like the right thing to do. Need some time to try this.

kung-foo, 2019-04-08:
Do any of the other wire protocols allow for `nil`?

wmnsk, 2019-04-08:
I think `nil` is useful to tell the difference between the cases "the field itself is missing" and "the field exists but the value is empty".

Can the current implementation of codec handle that difference in any way?

magiconair, 2019-04-08:
@kung-foo Regarding other protocols I don't know but I'll doubt it. I'll look though or if you want to please do.

@wmnsk The point is that the decoder will always create an empty struct and it has to because there is no way to express "not set" in the protocol. Otherwise, you have to have `nil` checks everywhere. Requiring the empty structs makes the protocol less robust since it is easier for the caller to generate wrong messages. I've been bitten by this now a number of times and I think having `&ua.LocalizedText{}` or `&ua.DiagnosticInfo{}` values does not help. 

Yes, we can make the encoder smarter. This was not hard but it broke all the tests which I could also work around by making the test runner take this into account. But IMO the underlying issue is that the protocol does not support `nil` for struct fields.

kung-foo, 2019-04-08:
At least from the encoder side, if we were to implement the `SOAP` transport, it seems like the std lib's XML encoder will not encode a `nil` struct. Versus a zero values that will be sent as an empty XML node.
For example: https://play.golang.org/p/HRjKvKJv08p

So the question is whether OPC/UA over `SOAP` requires all values to be expressed, or if the `omitempty` (from json/xml) concept can be used.

As I write this, I wonder if we should have a custom `opc` _tag_  (see: https://medium.com/golangspec/tags-in-golang-3e5db0b8ef3e) that can assist in the encoding step...

wmnsk, 2019-04-08:
@magiconair 
I got your point. I think making it work stable is the first priority so I agree on trying your idea to find the better way.
> not to generate struct fields as pointers since they cannot be nil anyway. Then the codec just works.

For the struct tags I sometimes use it in other protocols in my company. It's magical but tends to make it so complicated... should be used with much considerations. (no idea how we use it in this case at this moment, though)

magiconair, 2019-04-08:
We can add tags if we need to but I don’t think it solves anything for this specific problem. Why don’t we all do some digging? I’ll spend a little bit of time implementing this to get a feel for the change. 

Also, is anybody actually using the soap and json protocols?

wmnsk, 2019-04-08:
> Also, is anybody actually using the soap and json protocols?

For me I’ve never seen that but I don’t have such a long experience in the industry...
At least I can put +1 to implement the binary first until we can make it production ready.

magiconair, 2019-04-09:
Dropping all pointers was not a lot of work but the codec does not recognize the `BinaryEncoder` interface anymore. Also, `NodeID`, `Variant` and `GUID` need an `IsZero()` method which also bubbles up into the client code when you get a `Variant` as a result.

The other thought I had was that by switching to value semantics we have to handle more allocations since objects are now passed by value, AFAIR.

While value semantics feels right for the protocol itself I'm now wondering whether this is an implementation detail which bubbles up...

magiconair, 2019-04-09:
I've also been playing with the initial approach of making the encoder smarter since that is a much smaller change and I think @kung-foo has a point in using tags. They make the behavior explicit instead of relying on a heuristic which isn't necessarily obvious.

kung-foo, 2019-04-11:
yeah, while I am generally not a fan of annotations, I am also not a fan of heuristics (magic). So keeping any encoding-specific metadata as close to the struct possible seems like a good approach.

magiconair, 2019-04-20:
Lets park this. Feels like solving a non-issue. Lets re-open if someone feels strongly about this.

closed 2019-04-20 by magiconair