gopcua/opcua #167: Build in 32 bit environment fails

y94anden, 2019-04-16:
When crosscompiling for a 32 bit environment, I get the following two issues:
```
.../src/github.com/gopcua/opcua/ua/node_id.go:133:22: constant 4294967295 overflows int
.../src/github.com/gopcua/opcua/ua/node_id.go:250:10: constant 4294967295 overflows int
```

The first issue arises since `strconv.Atoi` a few lines earlier returns an `int` which in my system is 32 bit and thus cannot be compared to an `uint32`. Using `ParseUint` instead solved the problem:
```go
id, err := strconv.ParseUint(idval[2:], 10, 32)
```

The second issue is similar, but for that I made an uglier solution to get it to work by changing the input argument of SetIntID to an `uint32` and fiddling a bit with casting etc in the code below.

This relates to commit b9119dc of `ua/node_id.go`

magiconair, 2019-04-16:
@y94anden I've merged a change that should address this. Let us know if there are other issues.

closed 2019-04-16 by magiconair

y94anden, 2019-04-16:
Great, thanks! I will test some more tomorrow.

commit 959309d5 (2019-04-16): make NodeID numeric values uint32

Make sure that NodeID is strict about uint32.

Fixes #167
PR #168 (2019-04-16): Issue #167: 32bit build fails
Fix code for 32 and 64 bit platforms

Fixes #167