gopcua/opcua #170: RequestedSessionTimeout not set in func (c *Client) CreateSession

maurocordioli, 2019-04-19:
I was tring  to connect to an Siemens S7, using the datetime example I was unable to create a session.. 

the error returned was `StatusBadSessionIDInvalid`. 
so logging the tcp packets with wireshark and matching the results with a working client I noticed that the message `CreateSessionRequest `has a
`opcua.RequestedSessionTimeout = 0 `
so the S7 replied with 
`aopcua.RevisedSessionTimeout=1`

when at next step we call `ActivateSession `the session is already expired.

in the source code `client.go` I noticed that   in 
`func (c *Client) CreateSession` the `RequestedSessionTimeout `of the struct `ua.CreateSessionRequest` is not set  so it has 0 value.

for me worked adding `RequestedSessionTimeout: cfg.SessionTimeout`
 ```
req := &ua.CreateSessionRequest{

		ClientDescription:       cfg.ClientDescription,

		EndpointURL:             c.endpointURL,

		SessionName:             fmt.Sprintf("gopcua-%d", time.Now().UnixNano()),

		ClientNonce:             nonce,

		ClientCertificate:       c.cfg.Certificate,

		RequestedSessionTimeout: cfg.SessionTimeout,
	}
```

magiconair, 2019-04-19:
We should probably set a sensible default in the config. 

—
Frank Schröder

magiconair, 2019-04-19:
Hi @maurocordioli, could you see whether the patch in #171 works for you? I am wondering what a sensible default session timeout is? I've set it to 30 seconds for now but am happy to learn more about this. Any input on this @dwhutchison , @wmnsk or @kung-foo?

dwhutchison, 2019-04-19:
30s seems appropriate; I can't find a recommendation in the spec and the servers I've been testing against respond with RevisedSessionTimeouts ranging from 100ms to 15s.

maurocordioli, 2019-04-20:
> Hi @maurocordioli, could you see whether the patch in #171 works for you?
yes it works with a siemens S7 (CPU 1512SP F-1 PN)

maurocordioli, 2019-04-20:
bye the way , on the opc foundation test cases  they have a test case 006  when the RequestedSessionTimeout is too long ( a week) the server concedes a RevisedSessionTimeout of 60 seconds.also if client request a unresonable short  session timeout case 005 the server revise a value of 60 seconds.   may be 60 seconds could be a Ansatz reasonable timeout?   [https://apps.opcfoundation.org/ProfileReporting/testcases/Client/SessionClientGeneralServiceBehaviour.html](url)

magiconair, 2019-04-20:
60 seconds it is then.

PR #171 (2019-04-20): Honor session timeout
This patch honors the session timeout when creating a
session. It also changes the type of the SessionTimeout
from `float64` to a `time.Duration` and sets the default to 
60 seconds.

Fixes #170
commit 9db74f4a (2019-04-20): Issue #170: honor session timeout

This patch honors the session timeout when creating a
session. It also changes the type of the SessionTimeout
from float64 to a time.Duration.

Fixes #170
closed 2019-04-20 by magiconair