gopcua/opcua #172: Discussion on compliance tests

magiconair, 2019-04-20:


maurocordioli, 2019-04-20:
the opcua lib for rust uses a 3rd-party server as validation 
they are using node js node-opcua implementation.  https://github.com/locka99/opcua/blob/master/docs/setup.md#3rd-party-servers  
and a travis CI pipe line for automated tests. 
a generic image in travis has support for both go and node_js , 
so a test could spin up the node_js server opcua and run the golang tests

dwhutchison, 2019-05-27:
In https://github.com/gopcua/opcua/tree/feature/server I've started a uatest package (shamelessly inspired by the httptest package) for this.  A few dummy tests are in the main package under `client_test.go` to show what I'm thinking.  If you pass the `OPC_EXTERNALSERVER` environment variable with another server's endpoint, it'll forego starting up our server and clients can connect to another endpoint without changing the test code (eg `OPC_EXTERNALSERVER="opc.tcp://localhost:62541" go test`).  

There's lots more to be added (an inspector object like the `ResponseRecorder` and I'm hoping for a mechanism to override stack-controlled parts of the messages to mimic bad behaviour).

I think we could implement tests for as many of the published checks as possible.

Is this the right direction?

magiconair, 2019-05-27:
I think for the initial set of tests I would keep it even simpler. The only thing we really need is to start a server with a given behavior. The hard part is not the test server but the handler the server executes. We can either inline this or have test servers in `.js` files. Then starting the other server could be as simple as `exec.Command("node", "server.js", args...)` since we need different behavior of the server for each test.

magiconair, 2019-05-27:
OTOH, when I look at how complex it is to just create a simple server with node-opcua I'm inclined to just use https://github.com/open62541/open62541 for running a server which exposes a node so that we can read, write, browse and subscribe. We can embed an amalgamation of the .c/.h file and only need a C compiler. When you strip out all the comments then this is their simplest server.

https://github.com/open62541/open62541/blob/master/examples/tutorial_server_firststeps.c#L23-L47

bbergshaven, 2019-08-29:
Hello. I would love to contribute to this project. We have written to you on the slack channel as well. I have bought the official testing framework for opc-ua and can run tests on that as well.

magiconair, 2019-08-29:
Hi @bbergshaven, this is great. I'm not sure I can help you with where to start but if you need pointers on finding your way around I'm there to help. What would you like to work on?

Also, I'm not sure anyone is monitoring the Slack channel. Let's try to keep discussions on Github. If things become too detailed or I need to coordinate with another developer I've used Keybase but would like to keep this to a minimum. Things should be open but productive.