gopcua/opcua #180: There are some issues when run examples as default config

769139671, 2019-05-06:
when we config our opcua server and run examples as default config, we should change fllow func, we should delete :
                opt = AuthPolicyID("Anonymous")
		opt(c.cfg, c.sessionCfg)
if not, the server will return : The user identity token is not valid.

func NewClient(endpoint string, opts ...Option) *Client {
	c := &Client{
		endpointURL: endpoint,
		cfg:         DefaultClientConfig(),
		sessionCfg:  DefaultSessionConfig(),
	}
	for _, opt := range opts {
		opt(c.cfg, c.sessionCfg)
	}

	// UserIdentityToken was removed from DefaultSessionConfig() so ensure a default still is set
	if c.sessionCfg.UserIdentityToken == nil {
		opt := AuthAnonymous()
		opt(c.cfg, c.sessionCfg)
		opt = AuthPolicyID("Anonymous")
		opt(c.cfg, c.sessionCfg)
	}
	return c
}

wmnsk, 2019-05-06:
Hi,

The parameters an OPC UA server accepts vary much depending on implementations by vendors and/or configurations by users, and that's why we provide options feature in `NewClient()`. Please consider using `Option` funcs defined in `config.go`.

Are you sure your server worked if you remove these two lines from `NewClient()` in `client.go`?

```go
	opt = AuthPolicyID("Anonymous")
	opt(c.cfg, c.sessionCfg)
```

Then modify example itself instead of the codes in package; the following code works as the same.

```go
func main() {
	// ...

	// set an option not to hit the "c.sessionCfg.UserIdentityToken == nil" condition in NewClient()
	c := opcua.NewClient(*endpoint, opcua.AuthAnonymous())
	if err := c.Connect(); err != nil {
		log.Fatal(err)
	}
```

Hope this helps.

769139671, 2019-05-06:
Hello:
This works when add opcua.AuthAnonymous() to opcua.NewClient. 
My opc server also works when I remove these two lines from NewClient() using: 
func main() {
	// ...

	// set an option not to hit the "c.sessionCfg.UserIdentityToken == nil" condition in NewClient()
	c := opcua.NewClient(*endpoint)
	if err := c.Connect(); err != nil {
		log.Fatal(err)
	}

	// ...

}
So I suggest to change the examples, because it really confuse novices
                                                                                                      Thank you for your help!

wmnsk, 2019-05-06:
Nice to hear it worked! Could you please tell us which vendor's OPC UA server did you use? It is important for us to gather the interoperability information for considering how the default configuration should be.

FYI, It is one of Go's conventional way to handle options in that way. Hope these articles help you get familiar with it.

* https://commandcenter.blogspot.com/2014/01/self-referential-functions-and-design.html
* https://dave.cheney.net/2014/10/17/functional-options-for-friendly-apis

For the example programs, I think it'd be better to be kept as simple as possible because rich examples might look worse(not so easy to understand, etc.) for some people. The full feature should be well-documented in GoDoc instead, which I've just started working on.

769139671, 2019-05-06:
My OPC UA server is :  Prosys  OPC UA Simulation Server
thank you for your contributions!

magiconair, 2019-05-06:
@dwhutchison could you have a look at this, please? I think this touches on your recent changes.

magiconair, 2019-05-06:
The examples should work out of the box with servers in the default configuration.

769139671, 2019-05-06:
@magiconair It works well when add an arguments of opcua.AuthAnonymous() to NewClient().  It is not necessary to remove these two lines in NewClient(). But now the NewClient().   in examples only have one arguments, so that the server returns a wrong message.
Thank you!

dwhutchison, 2019-05-06:
Sure, I'll work on a change.  

The underlying problem is that some servers are very particular with the value of the Policy ID whereas others are lenient and will accept any value of Policy ID for Anonymous auth.  For example, I've tested servers that report their auth policy ID as 'anonymous' but won't accept 'Anonymous' as a valid value.  The correct values can always be obtained from the discovery services (eg. GetEndpoints).

769139671, 2019-05-07:
How to list out all available nodes bellow a folder of OPC UA server? is there any examples?
please

magiconair, 2019-05-07:
I have some code for browsing but let’s keep this issue about auth. Your questions are important since others will have similar issues. Can you open another issue for browsing, please?

dwhutchison, 2019-05-07:
@769139671 would you mind testing the PR I opened for this and let me know if it works for you?  I’ve tested against Prosys and a few others but I want to make sure I’m not overlooking a nuance in your setup.

769139671, 2019-05-08:
@dwhutchison I have tested your branch(fix/issue180-bad-policyID), it do not return err: "The user identity token is not valid." anymore.  When I run datetime.go in examples, it works well, but when I run read.go in examples  it print nothing finally with no error happened.

dwhutchison, 2019-05-08:
That appears to be unrelated to the original issue but I can reproduce; moved this one into #187

commit 668f6193 (2019-05-08): automatically set the anonymous auth policy ID when none was supplied by the user

fixes #180
closed 2019-05-08 by magiconair