gopcua/opcua #183: How to browse all available node?

769139671, 2019-05-07:
How to list out all available nodes bellow a folder of OPC UA server? 
@magiconair    please : )

magiconair, 2019-05-07:
Thank you and sorry for being such a stickler. I sincerely believe these tickets to be of value to others.

You can find the code in the `more-browsing` branch. I've written this for my own purpose so far. Let me know whether it works for you so that we can generalize it.

769139671, 2019-05-07:
I ran the more-browsing branch, it returns : browse children: Decoding halted because of invalid data in the stream. StatusBadDecodingError (0x80070000).
because the Prosys Opc UA server have a kind of data type named "Double",

closed 2019-05-14 by 769139671

jjauzion, 2022-01-13:
@magiconair I have the same question as cngobd, seems the branch `more-browsing` doesn't exist anymore. Any help would be much appreciated ! Thx :)
(see also https://github.com/gopcua/opcua/discussions/545)

peng110653, 2022-01-21:
https://github.com/gopcua/opcua/issues/183#issuecomment-1012384459
I have the same problem, @jjauzion    Have you solved your problem yet?