gopcua/opcua #185: I get ua.BrowseResponse on ua.PublishRequest

wsnotify, 2019-05-07:
Goroutine 1: Sending PublishRequest and waiting for response
Goroutine 2: Sending BrowseRequest and waiting for response
Goroutine 1: Received ua.BrowseResponse
Goroutine 2: waiting

Seems like related to SequenceNumber in #184

magiconair, 2019-05-08:
Can you check whether you still see that with the fix from #185?

wsnotify, 2019-05-13:
Yeah, still getting `invalid response: *ua.BrowseResponse`

alexbrdn, 2019-05-16:
I also get a response of a wrong type from a different goroutine with the current master

magiconair, 2019-05-16:
Can you try running this with `OPCUA_DEBUG="debug packet"` ?

magiconair, 2019-05-16:
This might provide some additional info on where to look or where we should add more debug messages.

alexbrdn, 2019-05-16:
I think there's no need for that. I believe I have found the culprit. Let me prepare a Merge Request and we'll discuss it there.

magiconair, 2019-05-16:
@wsnotify can you verify that this fixes your problem as well?

closed 2019-05-16 by magiconair

wsnotify, 2019-05-17:
@magiconair Looks like it helped.