gopcua/opcua #187: examples/read.go outputs blank line

dwhutchison, 2019-05-08:
From #180 by @769139671
>  When I run datetime.go in examples, it works well, but when I run read.go in examples it print nothing finally with no error happened.

The difference between the two examples is that `c.Read(req)` in read.go returns a `*ua.ReadResponse` but `c.Node()` in datetime.go returns a `*ua.Variant`.  Both have a .Value field, but the `*ua.ReadResponse`'s .Value is itself a `*ua.Variant`.  The .String() method on ua.Variant doesn't have conditions for all datatypes and defaults to "" for unhandled types (see ua/variant.go)

in examples/read.go: changing 
`log.Print(resp.Results[0].Value)` to `log.Print(resp.Results[0].Value.Value)` fixes this for me.

@magiconair, you're more familiar with this code; thoughts on a clean solution?  I'm not a fan of the .Value stutter but adding a condition for each subtype in the Variant .String() method doesn't sound great, either.  Do/could the subtypes have their own reasonable .String() methods so we could uncomment `return fmt.Sprintf("%v", m.Value)`?

769139671, 2019-05-08:
@dwhutchison It works well after changing now,  thank you very much!

magiconair, 2019-05-08:
Yeah, I’m aware. The current solution isn’t good enough. I’ll think about it.

magiconair, 2025-04-15:
Closing because of age. Re-open if necessary

closed 2025-04-15 by magiconair