gopcua/opcua #190: How to connect to the server with username & password?

769139671, 2019-05-09:
I want to connect to my Prosys OpcUA server with User Authentication Methods: Username & Passwords, and with the Security Modes: None ,
c:=opcua.NewClient(*endpoint,opcua.AuthUsername("yc","abc"),opcua.SecurityFromEndpoint())
	if err := c.Connect(); err != nil {
		fmt.Println("err mark 1")
		log.Fatal(err)
	} else {
		fmt.Println("connect success")
	}
	defer c.Close()
How to config "opcua.SecurityFromEndpoint()"? it needs two arguments
please

dwhutchison, 2019-05-09:
Have a look at the crypto example.  While it’s a bit verbose, it contains the steps necessary to connect to your server with SecPolicy None and AuthMode UserName. 

The steps involved would be: 
- load a client certificate (this shouldn’t be necessary with sec mode None; it’s on my radar to allow UserName auth without providing a cert). 
- get a list of the server endpoints from GetEndpoints()
- loop through the endpoints and find the endpoint you wish to use based on your sec policy/mode (None/None)
- provide that Endpoint to SecurityFromEndpoint with the second parameter as ua.UserTokenTypeUserName
- Provide the AuthUserName() option as you are

That should configure the client as you’re expecting. 

Note you can cache the results of GetEndpoints as they rarely change. This could save the request on each connect.

769139671, 2019-05-09:
@dwhutchison  The first parameter of  opcua.SecurityFromEndpoint() in opcua.NewClient() is *ua.EndpointDescription,  I configured it like this:
/*
var ep ua.EndpointDescription
	ep.EndpointURL = "opc.tcp://localhost:4840"
	ep.SecurityMode = ua.MessageSecurityModeNone
	ep.SecurityPolicyURI = "http://opcfoundation.org/UA/SecurityPolicy#None"

c := opcua.NewClient(*endpoint, opcua.AuthUsername("yc", "abc123"),
opcua.SecurityFromEndpoint(&ep, ua.UserTokenTypeUserName))
/
is that right? I ran it but it ruturns : The user identity token is valid but the server has rejected it. StatusBadIdentityTokenRejected (0x80210000)
please

769139671, 2019-05-09:
@dwhutchison I have found what's wrong with my code.  I should run crypro.go and generate a certification first, then install the certification in the Prosys Opcua server. finally, change the configurations in the crypto.go and make sure the flag value of gencert is false, it works well.  I think it is also useful for other novices
thank you for your help

closed 2019-05-09 by 769139671

piperoc, 2023-08-17:
I tried this kind of approach (no encryption but user/pwd authentication). 

Here's a code section:

```go
package main

import (
	"context"
	"flag"
	"fmt"
	"github.com/gopcua/opcua"
	"github.com/gopcua/opcua/debug"
	"github.com/gopcua/opcua/ua"
	"log"
	"os"
	"time"
)

func main() {

	//arguments
	endpointUrl := os.Args[1]
	progName := os.Args[2]

	if progName == "" || endpointUrl == "" {
		fmt.Println("Please provide a url and a test program name ...")
		return
	}

	var (
		endpoint = flag.String("endpoint", endpointUrl, "OPC UA Endpoint URL")
	)

	//create an endpoint description with the url, security mode none and security policy none
	var ep ua.EndpointDescription
	ep.EndpointURL = endpointUrl
	ep.SecurityMode = ua.MessageSecurityModeNone
	ep.SecurityPolicyURI = ua.SecurityPolicyURINone

	//create a slice of opcua options with user token type username and user name and password
	opts := []opcua.Option{
		opcua.SecurityFromEndpoint(&ep, ua.UserTokenTypeUserName),
		opcua.SessionTimeout(30 * time.Minute),
		opcua.AutoReconnect(true),
		opcua.ReconnectInterval(time.Second * 10),
		opcua.Lifetime(30 * time.Minute),
		opcua.RequestTimeout(3 * time.Second),
		opcua.AuthUsername("john", "master"),
	}

	flag.BoolVar(&debug.Enable, "debug", false, "enable debug logging")
	flag.Parse()
	log.SetFlags(0)

	ctx := context.Background()

	c := opcua.NewClient(*endpoint, opts...)
	if err := c.Connect(ctx); err != nil {
		fmt.Println("Connection Failed")
		log.Fatal(err)
	} else {
		fmt.Println("connect success")
	}
	defer c.Close(ctx)

        //do something
}

```

Testing against the UAAnsiCServer, getting the following message:
```
non-username authentication already configured, ignoring
connect success
```

I'm not sure where it defaulted to anonymous....
Thanks