gopcua/opcua #191: How to browse folder's in a NodeClass?

769139671, 2019-05-09:
@magiconair I want to browse all available folder's  belong to  a  NodeClass,
my NodeClass index in Prosys Opc UA simulation server :
NodeClass :
       NamespaceIndex:       0
       NamespaceURI:          http://opcfoundation.org/UA/
       IdentifierType :            Numeric
       Identifier:                    85
How to config the browse.go in examples.  Please?

magiconair, 2019-05-09:
The example in `browse.go` is a starting point but not a generic example for browsing since the OPC/UA data model is too complex. You can use the recursive `browseChildren` function as a starting point but how you filter and what you output depends on your specific use case.

769139671, 2019-05-10:
@magiconair Thank you for your help, now I want to browse all folders organized by root when I connect to the Prosys OpcUA server , is there any functions can implement it?
please

magiconair, 2019-05-10:
The code in `browse.go` should be a starting point but you have to find our yourself on how to do this. I don't have an example for this.

769139671, 2019-05-13:
@magiconair which functions should I use to get the "user access level" of a node?
Please

magiconair, 2019-05-16:
@769139671 I've created a PR with the methods for `AccessLevel` and `UserAccessLevel`. I'll wait for some comments from the other developers but then we can merge this.

769139671, 2019-05-20:
@magiconair Thank you!

closed 2019-05-20 by 769139671

magiconair, 2019-05-20:
I've just merged #199