gopcua/opcua #202: Examples error

wsnotify, 2019-05-17:
Trying to run any examples, getting this errors:
```bash
(in /github.com/gopcua/opcua/examples/read)
# go run ./read.go 
# github.com/gopcua/opcua/uapolicy
../../uapolicy/crypto_pkcs1v15.go:26:37: c.PrivateKey.PublicKey.Size undefined (type rsa.PublicKey has no field or method Size)
../../uapolicy/crypto_pkcs1v15.go:54:25: c.PublicKey.Size undefined (type *rsa.PublicKey has no field or method Size)
../../uapolicy/crypto_rsaoaep.go:32:37: a.PrivateKey.PublicKey.Size undefined (type rsa.PublicKey has no field or method Size)
../../uapolicy/crypto_rsaoaep.go:68:25: a.PublicKey.Size undefined (type *rsa.PublicKey has no field or method Size)
../../uapolicy/policyAes128Sha256RsaOaep.go:87:43: localKey.PublicKey.Size undefined (type rsa.PublicKey has no field or method Size)
../../uapolicy/policyAes128Sha256RsaOaep.go:88:142: localKey.PublicKey.Size undefined (type rsa.PublicKey has no field or method Size)
../../uapolicy/policyAes128Sha256RsaOaep.go:92:35: remoteKey.Size undefined (type *rsa.PublicKey has no field or method Size)
../../uapolicy/policyAes128Sha256RsaOaep.go:93:134: remoteKey.Size undefined (type *rsa.PublicKey has no field or method Size)
../../uapolicy/policyAes128Sha256RsaOaep.go:98:33: remoteKey.Size undefined (type *rsa.PublicKey has no field or method Size)
../../uapolicy/policyAes128Sha256RsaOaep.go:99:33: remoteKey.Size undefined (type *rsa.PublicKey has no field or method Size)
../../uapolicy/policyAes128Sha256RsaOaep.go:99:33: too many errors
```

magiconair, 2019-05-17:
You need to use go1.11 or higher since we're using modules. Use go1.12.5

magiconair, 2019-05-17:
I'll update the README

dwhutchison, 2019-05-17:
The .Size method was introduced in Go 1.11 so you’d need to upgrade. Is that an option for you?

@magiconair thoughts on supporting older versions?  This particular issue is easy to fix by including our own version of .Size(). I’m not sure what else needs newer go versions, though.

wsnotify, 2019-05-17:
Tried golang 1.11 (docker), same errors.

magiconair, 2019-05-17:
I think supporting the current and previous go versions as originally suggested by @wmnsk is a reasonable approach unless there is a good reason not to.

magiconair, 2019-05-27:
I'll close this. Feel free to comment/re-open if there are other issues.

closed 2019-05-27 by magiconair