gopcua/opcua #209: How to keep the session connect alive all the time?

769139671, 2019-05-26:
@magiconair I want to keep the session alive until I close the connect of the session by myself,where to config when connect to the opc server?

magiconair, 2019-05-26:
I'm not sure what you mean. Unless you close the client the session is not closed. If you want to transfer a session from one client to another you can use the `DetachSession` and `AttachSession` methods. You can find an example in `examples/crypto/crypto.go`.

769139671, 2019-05-29:
@magiconair  I have found what's wrong with this issue. If I use the DefaultSessionConfig to connect to opcServer,  the session will break off one minute later, it is because of DefaultSessinConfig : SessionTimeout: time.Minute.  If we change SessionTimeout longer, the session will reatain longer
Thank you for your attention to this issue

// DefaultSessionConfig returns the default configuration for a client
// to establish a session.
func DefaultSessionConfig() *uasc.SessionConfig {
	return &uasc.SessionConfig{
		SessionTimeout: time.Minute,
		ClientDescription: &ua.ApplicationDescription{
			ApplicationURI:  "urn:gopcua:client",
			ProductURI:      "urn:gopcua",
			ApplicationName: &ua.LocalizedText{Text: "gopcua - OPC UA implementation in Go"},
			ApplicationType: ua.ApplicationTypeClient,
		},
		LocaleIDs:          []string{"en-us"},
		UserTokenSignature: &ua.SignatureData{},
	}
}

magiconair, 2019-05-29:
No problem. Closing.

closed 2019-05-29 by magiconair