gopcua/opcua #212: We should start tagging

magiconair, 2019-05-27:
Even with code moving around and APIs changing I feel that we should start tagging with v0.0.x since this project is getting some traction. Then users can pin to specific versions. 

There will still be a stern warning that things can change at any time.

Any arguments against this?

dwhutchison, 2019-05-27:
Agreed that it'll help with identifying issues.  I think we'll just need some rules on how to tag changes and what kind of change qualifies for a version bump (vs being rolled into a later version)

magiconair, 2019-05-27:
My initial thought was to just tag everything with v0.0.xxx until we're comfortable calling it at least v0.1.0. OTOH, the client is already useful in its current form and we're finding and fixing bugs quite quickly. Terraform just release 0.12.0 as a major release for a million users. Maybe we shouldn't overthink it. 

Lets call it 0.1.0 and we keep bumping the patch version until we feel that it warrants a 0.2.0. Since we're all more or less volunteers here it doesn't make sense at this point to come up with release plans. Once we get a reasonable number of issues that we can work on we can label them for the next release.

I've written a small tool github.com/magiconair/bump which assists with signed tags. So we can just run `bump tag && git push --tags`.

OK?

magiconair, 2019-06-12:
Just tagged `master` with `v0.1.0`

closed 2019-06-12 by magiconair