gopcua/opcua #214: Basic128Rsa15+Sign+Anonymous requires private key

magiconair, 2019-06-04:
Our Siemens SIMATIC S7-1500 got reconfigured and anonymous access disabled. Minimum requirement is `Basic128Rsa15+Sign`. Running the `datetime` example unmodified results in a panic:

```sh
$ go run examples/datetime/datetime.go  -endpoint opc.tcp://192.168.118.30:4840
failed to send close secure channel request
panic: close of closed channel

goroutine 34 [running]:
github.com/gopcua/opcua/uasc.(*SecureChannel).Close(0xc0000f2050, 0x1226940, 0xc0000a0140)
	/Users/frank/src/github.com/gopcua/opcua/uasc/secure_channel.go:114 +0x49
github.com/gopcua/opcua/uasc.(*SecureChannel).recv(0xc0000f2050)
	/Users/frank/src/github.com/gopcua/opcua/uasc/secure_channel.go:351 +0xfe0
created by github.com/gopcua/opcua/uasc.(*SecureChannel).Open
	/Users/frank/src/github.com/gopcua/opcua/uasc/secure_channel.go:106 +0x3f
exit status 2
```

After updating the example with

```patch
$ git diff
diff --git a/examples/datetime/datetime.go b/examples/datetime/datetime.go
index 97f90ec..de5fa37 100644
--- a/examples/datetime/datetime.go
+++ b/examples/datetime/datetime.go
@@ -19,7 +19,12 @@ func main() {
        flag.Parse()
        log.SetFlags(0)

-       c := opcua.NewClient(*endpoint)
+       opts := []opcua.Option{
+               opcua.SecurityPolicy(ua.SecurityPolicyURIBasic128Rsa15),
+               opcua.SecurityMode(ua.MessageSecurityModeSign),
+               opcua.AuthAnonymous(),
+       }
+       c := opcua.NewClient(*endpoint, opts...)
        if err := c.Connect(); err != nil {
                log.Fatal(err)
        }
```

I have to provide a private key

```sh
$ go run examples/datetime/datetime.go  -endpoint opc.tcp://192.168.118.30:4840
invalid channel config: Security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' requires a private key
exit status 1
```

This is not required when using the Prosys client with the same server.

@dwhutchison any idea?

dwhutchison, 2019-06-04:
Sign mandates a client key so if SecurityMode None has been disabled on your PLC/Server, the key requirement is expected behavior. The key is required for sign because symmetrical keys need to be negotiated to sign messages so the asymmetrical portion of the channel creation is still fully encrypted.  The symmetrical parts just aren't encrypted after signing keys are exchanged.  I suspect the prosys client might be providing it's own key.

The panic before you changed the mode is concerning and I'll try reproduce that.  The non-crypto examples assume SecMode None and Anonymous auth but that's fine provided they error appropriately and don't panic.

dwhutchison, 2019-06-05:
I was able to reproduce it with the UnifiedAutomation AnsiC server.  This server (and I suspect the Siemens PLC, too), sends an ERRF response to the `CreateSessionRequest` with Error: `StatusBadSecurityPolicyRejected`.  This is in contrast to other servers I tested against (that don't panic) which either time out or respond with a `ServiceFault` message (With Error `BadSecurityPolicyRejected`)

When the ERRF is sent, we try to close the SecureChannel twice which results in trying to close the `close` channel twice
-  once in `recv()`:
```go
	if errf, ok := err.(*uacp.Error); ok {
		s.notifyCallers(errf)
		s.Close()
		continue
		...
```
- and again in the client `Connect` method after the recv() bubbles the error up:
```go
	s, err := c.CreateSession(c.sessionCfg)
	if err != nil {
		_ = c.Close() // This calls secChan.Close()
		return err
	}
	...
```

Simply removing the `s.Close()` in `recv()` closes the channel without a panic as expected.

Of note, #211 doesn't actually exhibit the panic because the `close` channel has been removed.  That said, the underlying problem is still there as it tries to close the connection twice, but the only consequence is an extra error message in the debug messages.  

If there are no issues with removing the `s.Close()` on ERRF, I can incorporate this fix into #211

magiconair, 2019-06-05:
I've pushed a one-line fix for the double close. Should have created a separate ticket from the start. See #215

magiconair, 2019-06-05:
I'm still trying to get the `datetime` example to work. I've added some config helpers in #216 to set the policy, mode and key. It is not clear to me whether I need to provide a cert and a key or just the key. In any case I'm getting the same error in both cases.

Still checking what has changed on the PLC.

```shell
$ go run examples/datetime/datetime.go -endpoint opc.tcp://192.168.118.30:4840 -policy Basic128Rsa15 -mode sign -key ~/opc-cert/192.168.254.17-urn_freeopcua_client-2048/key.pem   -debug
debug: Connect to opc.tcp://192.168.118.30:4840
debug: conn 1: start HEL/ACK handshake
debug: conn 1: sent HELF with 61 bytes
debug: conn 1: recv ACKF with 28 bytes
debug: conn 1: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0x2000, SendBufSize:0x2000, MaxMessageSize:0x200000, MaxChunkCount:0x100}
debug: conn 1: close
asn1: syntax error: sequence truncated
exit status 1
```

magiconair, 2019-06-05:
@dwhutchison do you have an idea what could cause this error message?

```
asn1: syntax error: sequence truncated
```

dwhutchison, 2019-06-05:
Probably a missing or malformed certificate.  ASN1 is the encoding scheme used in the certificates (both client and server)

magiconair, 2019-06-05:
> Oh, the issue is that you're not loading the server's certificate and this error is coming from
>
> func (s *SecureChannel) openSecureChannel() error {
>     ...
>     remoteCert, err := x509.ParseCertificate(s.cfg.RemoteCertificate)
> In the crypto example, we use the flags to identify which endpoint from GetEndpoints best matches what the user chose and then use SecurityFromEndpoint to populate all the necessary info from the server, including the certificate.
>
> The datetime example uses the flags to directly set the connection parameters but we don't have a way to get/set the server's certificate. Per the spec, the discovery services are the way to get this info and clients are expected to know the endpoint details before they make a real connection. I think that's why I never added a ServerCertificate() option method since I assumed if you had that info, you also had the full endpoint to give to SecurityFromEndpoint. I think adding a ServerCertificate() option to allow users to cache the cert and customize the connection parameters is a good idea to complete the config API. However, I see this as the 'Advanced Mode' version of configuring security and SecurityFromEndpoint as the easy mode.

That makes sense. I’ll add the config option and then switch to auto mode. 

Thanks

—
Frank Schröder

magiconair, 2019-06-05:
I'll try to improve the error message as well.

magiconair, 2019-06-05:
Figured it out.

magiconair, 2019-06-05:
I'll close this ticket and create a new one for the error message.

closed 2019-06-05 by magiconair