gopcua/opcua #218: Warn when trying to use Basic256Sha256 with a Sha1 certificate

magiconair, 2019-06-05:
SecurityPolicy Basic256Sha256 requires a client certificate with a SHA256 signature. A lot of openssl generated certificates have a SHA1 signature. The client should warn that the certificate does not match the security policy since that isn't obvious unless you know where to look.

dwhutchison, 2019-06-06:
I don't mind working on this if it's not urgent (been busier than usual lately).
The proper fix to this (and probably #217 by extension) is to implement a certificate validation routine.  It's required by spec and was on my radar anyway.  The idea would be to check the remote certificate for validity (trusted CA, expiration, revocation status, etc.) before connecting to a server so we could use the same routines to verify the client certificate.  In addition to the x509 checks we could include checks for policy constraints like this one.

magiconair, 2019-06-06:
Sure. Go ahead. 

—
Frank Schröder

closed 2025-04-29 by github-actions[bot]