gopcua/opcua #221: cannot run the example codes

Felikesw, 2019-06-12:
When I was trying out the example code: run examples/datetime/datetime.go -endpoint opc.tcp://localhost:4840 , the following error came up:

dial tcp 127.0.0.1:4840: connectex: No connection could be made because the target machine actively refused it.
exit status 1

What is could be the problem, and how could I fix it?

wmnsk, 2019-06-12:
That’s because you ran OPC-UA client while no server was running on your local machine.
Do you have your own OPC-UA server? Then please run the program with the hostname or IP address specified instead of ‘localhost`.

Felikesw, 2019-06-12:
Ah, I see, no I don't have one yet, I just started looking into this today. Thanks for the comment!

wmnsk, 2019-06-12:
OK. Thank you for getting interested in our project. I hope you find this useful. Don’t hesitate to let us know on another issue if any 😀

closed 2019-06-12 by wmnsk