gopcua/opcua #222: How many node can I read once?

769139671, 2019-06-18:
@magiconair I read a simulation server which have 6000 nodes,  and it return an err:   The size of the message specified in the header is too large. StatusBadTCPMessageTooLarge (0x80800000)
does that means the number of nodes I send to the server is out of range?

magiconair, 2019-06-18:
Can you enable debug output to see whether that comes from the server or the client? Check the examples on how to do this. 

My guess is that the total message size exceeds the limit of what the server expects but there is also a chance that there is a bug in our chunking code. 

—
Frank Schröder

769139671, 2019-06-19:
You are right, this error is comes from server.  Thank you for your help!  : )

closed 2019-06-19 by 769139671