gopcua/opcua #223: when I run the client and read nodes,it returned an error occasionally:

769139671, 2019-06-24:
wsasend: An existing connection was forcibly closed by the remote host.
and I can not repeat this error, I want to know what happend with this error, 
thank you.

dwhutchison, 2019-06-24:
Does your server have any logs hinting at why it would be closing the connection?

What function calls are you making when this occurs? How often does this occur?

If you find out how to reproduce it, please run the application with debug mode on and share the logs

769139671, 2019-06-25:
@dwhutchison I had reproduced this error, when I read tags once per second after connected to the server,   this error will occur about one hour later.
Sorry I can not get logs of server now, can you help me with this error?

jkissinger, 2019-07-15:
@769139671 
This is a network level error (for example, see: https://stackoverflow.com/questions/2582036/an-existing-connection-was-forcibly-closed-by-the-remote-host ).  That's not to say there is no bug in the gopcua library here, it's just not possible to determine the cause with the information you've provided.  You'd need server logs or application debug level logs like you were asked for above.

closed 2019-09-10 by 769139671