gopcua/opcua #236: How to access a node by StringNodeId ?

sdghchj, 2019-07-09:
Is there any way to access a node by path of DisplayName?

for example:

ua.NewStringNodeId(2,"demo_led.temperature")

[image]

magiconair, 2019-07-09:
We don’t have a method to find a node by display name. You can browse for the browse name but that is not as simple as using ns=2;i=2

—
Frank Schröder

magiconair, 2019-07-10:
I think adding support for `TranslateBrowsePathsToNodeIds` might address this. Do you want to send a PR?

sdghchj, 2019-07-11:
> I think adding support for `TranslateBrowsePathsToNodeIds` might address this. Do you want to send a PR?

Yes, check PR, please

PR #237 (2019-07-19): add method:translate relative path of browse name to node id
Closes #236
closed 2019-07-19 by magiconair