gopcua/opcua #241: Unable to write arrays due to StatusBadEndOfStream

jkissinger, 2019-07-17:
Below is the code I'm using:

`func arrayWriteTest(client *opcua.Client) error {
  variant, err := ua.NewVariant(int32(0))
  if err != nil {
    return err
  }
  variant.ArrayLength = 8
  variant.Value = []int32{0, 1, 2, 3, 4, 5, 6, 7}
  variant.EncodingMask = byte(variant.EncodingMask + ua.VariantArrayValues)
  wv := &ua.WriteValue{
    NodeID:      ua.NewStringNodeID(4, "my.node.id"),
    AttributeID: ua.AttributeIDValue,
    Value: &ua.DataValue{
      EncodingMask: ua.DataValueValue,
      Value:        variant,
    },
  }
  req := &ua.WriteRequest{
    NodesToWrite: []*ua.WriteValue{wv},
  }
  _, err = client.Write(req)
  return err
}`

Here's the logs:
>debug: uasc 1/4: send *ua.WriteRequest with 128 bytes
>debug: conn 1: recv ERRF with 16 bytes
>debug: uasc 1/4: Cannot move beyond end of the stream. StatusBadEndOfStream (0x80B00000)
>debug: Received error: Cannot move beyond end of the stream. StatusBadEndOfStream (0x80B00000)
>2019/07/17 16:30:07.972023 failed array write test: Cannot move beyond end of the stream. StatusBadEndOfStream (0x80B00000)

I'll try to get server logs tomorrow, unless someone is able to reproduce/debug before then.

magiconair, 2019-07-17:
Variants don't handle arrays properly. I'll fix this.

magiconair, 2019-07-19:
Tested #242 with a Python server and this modified test function:

```go
func arrayWriteTest(client *opcua.Client) error {
	variant, err := ua.NewVariant([]uint32{0, 1, 2, 3, 4, 5, 6, 7})
	if err != nil {
		return err
	}
	wv := &ua.WriteValue{
		NodeID:      ua.NewNumericNodeID(2, 3),
		AttributeID: ua.AttributeIDValue,
		Value: &ua.DataValue{
			EncodingMask: ua.DataValueValue,
			Value:        variant,
		},
	}
	req := &ua.WriteRequest{
		NodesToWrite: []*ua.WriteValue{wv},
	}
	_, err = client.Write(req)
	return err
}
```

Could you verify that the patch works for you as well?

magiconair, 2019-07-19:
This is the FreeOPCUA python server I've used:

```python
#!/usr/bin/env python3

import sys
sys.path.insert(0, "..")
from opcua import ua, Server

if __name__ == "__main__":
    server = Server()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

    # get Objects node, this is where we should put our nodes
    objects = server.get_objects_node()

    # populating our address space
    myobj = objects.add_object(idx, "MyObject")
    myvar = myobj.add_variable(idx, "MyVariable", 6.7)
    myvar.set_writable()
    myarr = myobj.add_variable(ua.NodeId(3, idx), "MyArray", ua.Variant([], ua.VariantType.UInt32))
    myarr.set_writable()
    server.start()
```

commit ffcf3d12 (2019-07-19): Add support for arrays in variants

This patch adds support for one and multi-dimensional arrays in Variant
types.

Fixes #241
commit aa69324c (2019-07-24): Add support for arrays in variants

This patch adds support for one and multi-dimensional arrays in Variant
types.

Fixes #241
PR #242 (2019-07-24): Add support for arrays in variants
This patch adds support for one and multi-dimensional arrays in Variant
types.

Fixes #241
closed 2019-07-24 by magiconair