gopcua/opcua #243: ua.Variant should implement MarshalJSON

kung-foo, 2019-07-19:
With #242 making `Variant`'s members private, we might want to add explicit support for serializing a value via JSON. Not sure if the OPC UA Foundation has a schema or if we want to create a new one. In the simplest case, the serialized output will be lossy (i.e. you can tell if `"value": 4` is a 16 or 32bit int).

magiconair, 2019-07-19:
Might also want to add that for NodeID.

sruehl, 2023-08-16:
@kung-foo what we did in [PLC4X](plc4x.apache.org) is to use additional [suffixed keys](https://github.com/apache/plc4x/blob/develop/plc4go/internal/s7/s7Io_test.go#L265) to transport this information e.g.:
```json
{
  "value": 4,
  "value_gopcua_type": "int",
  "value_gopcua_bitlength": "16"
}
```
Maybe this is a option.
I could also look into this issue in a PR to get a bit more common with the codebase of gopcua.

closed 2025-04-29 by github-actions[bot]