gopcua/opcua #254: Prosys Simulation Server returns duplicate subscription IDs

kung-foo, 2019-08-02:
When creating multiple subscriptions in quick succession, I have seen the same sub ID come back. This causes an overwrite here: https://github.com/gopcua/opcua/blob/v0.1.4/client.go#L556
Which then causes data change notifications to be dispatched to the wrong handler.

Quick fix is to check for an existing Subscription ID, and return an error. But since the subscription has already been created on the server, do we need to cleanup?

Or is this technically a valid response?

Log: [prosys.log](https://github.com/gopcua/opcua/files/3461440/prosys.log)

You can see two `CreateSubscriptionResponse` come back with `SubscriptionId=28`

kung-foo, 2019-08-02:
@alexbrdn any insight into this one?

alexbrdn, 2019-08-02:
I believe it's a bug in Prosys Simulation Server. Spec says the following about CreateSubscription Service:

> | Name | Type | Description |
> | --- | --- | --- |
> | Response | | |
> | subscriptionId | IntegerId | The Server-assigned identifier for the Subscription (see 7.14 for IntegerId definition). This identifier shall be unique for the entire Server, not just for the Session, in order to allow the Subscription to be transferred to another Session using the TransferSubscriptions service. After Server start-up the generation of subscriptionIds should start from a random IntegerId or continue from the point before the restart. |

[excerpt from Part 4 Section 5.13.2.2, Table 88 – CreateSubscription Service Parameters]

kung-foo, 2019-08-02:
OK, that's what I thought. I'll make a PR to protect against it.

closed 2019-08-07 by magiconair

kung-foo, 2019-08-13:
Prosys confirmed this a bug in their SDK/Simulator:

> This was a bug in the (Java) SDK. It is fixed in the SDK version 4.0.0 (https://downloads.prosysopc.com/opcua/Prosys_OPC_UA_SDK_for_Java_4_Release_Notes.html#version-4-0-0), but we have not yet released a version of the simulation server with SDK 4.x, when we do this issue will be fixed. Hopefully that should happen within few weeks.

via: https://forum.prosysopc.com/forum/opc-ua-simulation-server/duplicate-subscription-ids/

PR #255 (2019-08-07): check for invalid/duplicate subscription id
fixes #254