gopcua/opcua #262: Discussion on guma

smyrman, 2019-09-02:
We previously reached out on Slack, and where advised to move the discussion here.

> We want to reach out as we started writing our own OPC UA client-only library for Go back in 2017 (that we never quite finished), and have now opened up this code in case it could be of some inspiration: https://github.com/searis/guma

> From us briefly going through the code of gopcua, we think you have done a great job, and have gotten a lot further than we ever did. One concept we had that could perhaps be of some inspiration though, is that we generated the client-side code for calling OPC UA services: https://github.com/searis/guma/blob/master/stack/services_auto.go

@magiconair:

> As for generating the client-side code: How do you handle sane defaults and custom error logic for the different client calls? I’ve tried to make the client calls as simple as possible so that the boilerplate is kept to a minimum. (https://github.com/gopcua/opcua/blob/master/client.go#L501-L535)
Maybe you have an idea.

It's some years since I wrote the code, and we never used it in production, but I will answer to the best of my ability.

The generated service code in Guma (previously linked file) was meant as a low-level interface (for developing more higher level functions on top when necessary, in which we only did for the "dial" bit I believe). Therefore the generated service calls did not set any sane defaults. The only thing it really did was to connect the relevant call parameters to the relevant (generated) response structs as described in the XML spec for all services defined by the spec.

We haven't tested or used all the generated service calls; I believe we only got as far as to test the connection related calls as well as Browse and Read/Subscribe.

smyrman, 2019-09-02:
The template we used to generate services can be found here:
- https://github.com/searis/guma/blob/master/generate/cmd/opcua-xml2code/services.go

magiconair, 2019-09-02:
We're generating all the service objects and it wouldn't be much work to also generate the generic client layer. My gut feel is that at this point it does not add so much additional functionality since you're then either pushing the heavy lifting to the user or you need to build another function which adds the sane behavior.

Adding the service call without helpers and sanity checks is only a couple of lines. You can call into the client directly if needed. (We should probably export `safeAssign` to make this simpler). See here an example for the `Write` method: https://github.com/gopcua/opcua/blob/master/client.go#L501-L508

IMHO, The real value comes from the sane defaults, the error handling and the examples which cannot be generated and take more time to write. 

I know that a couple of libraries take this low-level/high-level client approach but I've found that confusing. The `Client` should be useful as is and you can break-out if needed. 

What do others think?

smyrman, 2019-09-02:
> My gut feel is that at this point it does not add so much additional functionality since you're then either pushing the heavy lifting to the user or you need to build another function which adds the sane behavior.

Indeed, you need another function on top, or more likely, I would split it into two APIs; low-level (but with basic type validatoin), and high-level for easy-usage.

A low level client could be generated exposing a type-safe way of calling all 40 or so service calls exactly as defined by the spec, including all the handshake calls, but with no sane defaults. This may be used by end-users for very advanced use-cases, or when a higher level API is either not available, or not sensible to implement.

For most use-cases, users would use your current (high-level) client/API. This API can diverge from the interface defined by the spec where it's useful, e.g. by creating more easy to use connection/dial setup, track subscriptions on the client, or implement `DeleteSubscription` as a `Delete` method on a `Subscription` representatoin object. Some methods may not make sense to expose at all. Some of these examples may be a bit off.

How much value you feel a low-level client would add, is up to you.

Our main motivation for reaching out, was to see if there is anything we did back in 2017 that could potentially be of some inspiration or use to you. If it's not, there is no loss to anyone 🙂

magiconair, 2019-09-10:
After thinking about this I think that our current approach is lean enough to add one feature at a time without giving the impression we support more features than we actually do. I'm also a bit reluctant to add another layer to the client since it does not seem to provide enough benefit. As long as users can construct their own requests the Client should be as useful as possible with sane default behavior. We can always revisit this if someone has a concrete solution.

closed 2019-09-10 by magiconair