gopcua/opcua #264: NodeID type

Talbourdet, 2019-09-05:
Hello,

I don't understand how to build my NodeID.

I have tested the sample code with `'ns=0;i=2261'`  for the ProductName and it's work.

In don't anderstand how to write my NodeID for the one below.

`NS4|String||var|WAGO 750-8102 PFC100 CS 2ETH RS.Application.PLC_PRG.testInt`

Could someone help me please?

dwhutchison, 2019-09-07:
The format is largely dependant on the OPCUA server and device driver you’re using, but from what you’ve provided I’d try:

ns=4;s=RS.Application.PLC_PRG.testInt

If that doesn’t work, let me know which server and device driver you’re using.

magiconair, 2019-09-26:
I'll close this for now. Please re-open if you have additional questions.

closed 2019-09-26 by magiconair