gopcua/opcua #267: OPC UA PubSub DataSetMetaData

joesan, 2019-09-10:
I'm following the implementation on the PubSub and I'm trying to understand the codebase in the extobjs_gen.go where the DataSetMetaDataType is defined. What I do not understand is that how it matches with the OPC UA PubSub specification in Part 14 on page 25, _Chapter 6.2.2.1.2 DataSetMetaDataType_ where it says in _Table 3 – DataSetMetaDataType Structure_ 

From that table I see only the following fields:

name
description
fields
dataSetClassId
configurationVersion

But in the DataSetMetaDataType defined in the extobjs_gen.go, I see the following additional fields:

```
Namespaces           []string
StructureDataTypes   []*StructureDescription
EnumDataTypes        []*EnumDescription
SimpleDataTypes      []*SimpleTypeDescription
```

What is the relation between these fields and the one defined in the Spec in the Table 3?

magiconair, 2019-09-10:
The types in `extobj_gen.go` are generated from `schema/Opc.Ua.Types.bsd`. There the `DataSetMetaDataType` is defined as follows:

```xml
  <opc:StructuredType Name="DataSetMetaDataType" BaseType="tns:DataTypeSchemaHeader">
    <opc:Field Name="NoOfNamespaces" TypeName="opc:Int32" />
    <opc:Field Name="Namespaces" TypeName="opc:String" LengthField="NoOfNamespaces" />
    <opc:Field Name="NoOfStructureDataTypes" TypeName="opc:Int32" />
    <opc:Field Name="StructureDataTypes" TypeName="tns:StructureDescription" LengthField="NoOfStructureDataTypes" />
    <opc:Field Name="NoOfEnumDataTypes" TypeName="opc:Int32" />
    <opc:Field Name="EnumDataTypes" TypeName="tns:EnumDescription" LengthField="NoOfEnumDataTypes" />
    <opc:Field Name="NoOfSimpleDataTypes" TypeName="opc:Int32" />
    <opc:Field Name="SimpleDataTypes" TypeName="tns:SimpleTypeDescription" LengthField="NoOfSimpleDataTypes" />
    <opc:Field Name="Name" TypeName="opc:String" />
    <opc:Field Name="Description" TypeName="ua:LocalizedText" />
    <opc:Field Name="NoOfFields" TypeName="opc:Int32" />
    <opc:Field Name="Fields" TypeName="tns:FieldMetaData" LengthField="NoOfFields" />
    <opc:Field Name="DataSetClassId" TypeName="opc:Guid" />
    <opc:Field Name="ConfigurationVersion" TypeName="tns:ConfigurationVersionDataType" />
  </opc:StructuredType>
```
https://github.com/gopcua/opcua/blob/master/schema/Opc.Ua.Types.bsd#L373-L388

which `cmd/schema/main.go` translates to the Go struct

```go
type DataSetMetaDataType struct {
	Namespaces           []string
	StructureDataTypes   []*StructureDescription
	EnumDataTypes        []*EnumDescription
	SimpleDataTypes      []*SimpleTypeDescription
	Name                 string
	Description          *LocalizedText
	Fields               []*FieldMetaData
	DataSetClassID       *GUID
	ConfigurationVersion *ConfigurationVersionDataType
}
```
https://github.com/gopcua/opcua/blob/master/ua/extobjs_gen.go#L102-L112

I see these fields also in my Part 14 document (OPC UA Part 14 - PubSub RC 1.04.24 Specification.docx)

[image]

joesan, 2019-09-11:
Ok may be that explains why! I guess you are referring to the Release Candidate version of the Spec., but I was referring to the actual release version of the Spec in which Table 3 is where DataSetMetaDataType is defined. 

The DataTypeSchemaHeader is then defined as a super type of the DataSetMetaDataType as it can be seen in the link below!

https://reference.opcfoundation.org/Core/DataTypes/DataTypeSchemaHeader/

joesan, 2019-09-11:
I will close this issue as I understood what I wanted!

closed 2019-09-11 by joesan