gopcua/opcua #269: Parsing DataSetMessage using DataSetMetaData

joesan, 2019-09-11:
I'm looking at the following function:
```

// IntID returns the identifier value if the type is
// TwoByte, FourByte or Numeric. For all other types IntID
// returns 0.
func (n *NodeID) IntID() uint32 {
	return n.nid
}
```

Which is defined in the node_id.go. Why is this function returning 0 for all the other types except TwoByte, FourByte and Numeric. The other ones like String, Guid, Opaque, how are they then supposed to be handled?

magiconair, 2019-09-11:
What should `IntID()` return if the NodeID is of type `String` or `GUID`?

joesan, 2019-09-11:
I get your point, but what will the caller of this function do with the result? Or rather my question was to understand where this function will be used and in which context?

magiconair, 2019-09-11:
You can use `Type()` to determine the type of the node id:

https://github.com/gopcua/opcua/blob/master/ua/node_id.go#L167-L170

magiconair, 2019-09-11:
That depends on your application. Maybe you store the node ids somewhere else and need this value to look it up. Right now `NodeID` is one of the data structures of the OPC/UA protocol and the function provides access to one of its values.

joesan, 2019-09-11:
One of my colleagues here at work was about implementing the OPC PubSub and he used goopcua library and in his code where he is parsing the DataSetMessage Delta frame using the DataSetMetaData, he does a check like this:

`if structureDataType.DataTypeID.Namespace() == fieldMeta.DataType.Namespace() && structureDataType.DataTypeID.IntID() == fieldMeta.DataType.IntID()`

I could not get the comparison with the IntID, hence I thought of checking this here!

kung-foo, 2019-09-11:
You could do a string compare on the nodes: `if structureDataType.DataTypeID.String() == fieldMeta.DataType.String() { ... }`

joesan, 2019-09-16:
I will close this issue as I understood what I wanted!

closed 2019-09-16 by joesan