gopcua/opcua #272: Parsing GUID From Input

joesan, 2019-09-16:
I'm looking at the GUID parser implementation and form what I read from the OPC UA Spec version 1.04, Part 5 Chapter 5.1.3, it says in Table 2 the following about GUID:

| Component  | Data Type |
| ------------- | ------------- |
| Data 1 | UInt32 |
| Data 2  | UInt16 |
| Data 3  | UInt16 |
| Data 4  | Byte [8] |

But when I look into how it is implemented in the source code, I see the following function in the datatypes.go:

```
// GUID represents GUID in binary stream. It is a 16-byte globally unique identifier.
//
// Specification: Part 6, 5.1.3
type GUID struct {
	Data1 uint32
	Data2 uint16
	Data3 uint16
	Data4 uint64
}
func (g *GUID) Decode(b []byte) (int, error) {
	buf := NewBuffer(b)
	g.Data1 = buf.ReadUint32()
	g.Data2 = buf.ReadUint16()
	g.Data3 = buf.ReadUint16()
	g.Data4 = buf.ReadUint64()
	return buf.Pos(), buf.Error()
}
```
Is this some bug? Clearly the Data 4 is being read as UInt64 while it has to be read just as a Byte!

magiconair, 2019-09-16:
`Byte [8]` is an array of 8 bytes which is the same size as a `uint64`

magiconair, 2019-09-16:
Also, if it would be just a single byte you would not end up with a 16 byte value:

4 + 2 + 2 + 1 = 9

joesan, 2019-09-16:
Ok I see it! I guess I mis-interpreted the Data 4 to be of Byte rather than a Byte array of 8 bytes!

closed 2019-09-16 by joesan

magiconair, 2019-09-16:
no problem