gopcua/opcua #276: StatusBadInvalidArgument (0x80AB0000) after trying to connect a client

Lynckx, 2019-09-20:
I am trying to connect to a remote server and i get a StatusBadInvalidArgument (0x80AB0000) error. Does anyone know what this could be?
```
debug: Connect to opc.tcp://opcuademo.server.com:4840
debug: conn 4: start HEL/ACK handshake
debug: conn 4: sent HELF with 75 bytes
debug: conn 4: recv ACKF with 28 bytes
debug: conn 4: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0xffff00, MaxChunkCount:0x100}
debug: uasc 4/2: send *ua.OpenSecureChannelRequest with 1503 bytes
debug: conn 4: recv ERRF with 16 bytes
debug: uasc 4/2: One or more arguments are invalid. StatusBadInvalidArgument (0x80AB0000)
debug: Received error: One or more arguments are invalid. StatusBadInvalidArgument (0x80AB0000)
debug: conn 4: close
```
(not actual endpoint in debug logs) This is after i have created a client, selected an endpoint and tried to connect the client.

magiconair, 2019-09-20:
Can you post a tcpdump? Do you have a tcpdump from a client that works?

Lynckx, 2019-09-20:
I do not have a tcp dump, how can i get one?

magiconair, 2019-09-20:
I'd start with WireShark. Select the interface and filter on `port 4840`. It should decode the OPC/UA packets. The difference between a client that works and ours should be easy to spot.

Lynckx, 2019-09-20:
[StatusBadInvalidArgument(0x80AB0000)_tcpdump.zip](https://github.com/gopcua/opcua/files/3635637/StatusBadInvalidArgument.0x80AB0000._tcpdump.zip)
Here is the tcp dump

magiconair, 2019-09-26:
Wonder where `OpenSecureChannel message: ServiceId 0` is coming from.

magiconair, 2019-09-26:
A pcapng file would be helpful. Maybe base64 encode it and share it as a gist? Can you also share the code that initiates the connection?

Lynckx, 2019-10-03:
[StatusBadInvalidArgument-0x80AB0000-examples-crypto.zip](https://github.com/gopcua/opcua/files/3685189/StatusBadInvalidArgument-0x80AB0000-examples-crypto.zip)
I am not sure what you mean with "base64 encode it and share it as a gist". I just tried to connect with the endpoint using the examples/crypto command and i got the same error message. See the pcapng file. I also contacted the owner of this opc ua server but he is currently unavailable.

dwhutchison, 2019-10-03:
I looked at the wireshark log and there's not much to see since the `OpenSecureChannel` message (the one causing the error) is encrypted.

I have a couple follow-up questions, though:
1. Are you using `-auth-mode UserName` when connecting to the server?  The endpoints show that your server only supports username auth (i.e. not Anonymous)
2. Have you trusted the client certificate on the server?

Lynckx, 2019-10-03:
@dwhutchison  
1. Yes
2. I was told the server automatically accepts any incoming certificates, The server is owned by someone else who set this up for demo purposes. I am trying to get in contact with him to double check this but he is currently unavailable.

closed 2025-04-29 by github-actions[bot]