gopcua/opcua #277: questions regarding extension objects

kung-foo, 2019-09-24:
Couple of extension object related questions

1) Is there any reason we don't register _all_ of them? Or are we expecting the user to register only what they need? For example, I need to add this to my code:

```go
func init() {
	ua.RegisterExtensionObject(ua.NewNumericNodeID(0, id.BuildInfo_Encoding_DefaultBinary), new(ua.BuildInfo))
}
```

2) I am wondering if we can make this copy-pasta-error-prone-code at little bit less verbose:

```go
node := client.Node(ua.NewNumericNodeID(0, id.Server_ServerStatus_BuildInfo))

v, err := node.Value()
if err != nil {
	return err
}

if eo, ok := v.Value().(*ua.ExtensionObject); ok && eo.TypeID.NodeID.IntID() == id.BuildInfo_Encoding_DefaultBinary {
	bi := eo.Value.(*ua.BuildInfo)
	...
}
```

magiconair, 2019-09-24:
1. No, PR welcome
2. same, always open for suggestions

complex code is cause for errors

magiconair, 2019-09-24:
as an aside I also would like to merge the `id` package with the `ua` package and generate some better names.

magiconair, 2020-07-23:
@kung-foo is there a list of all the standard extension objects? When I go through the list of IDs I see standard request/response objects with extension objects.

kung-foo, 2020-07-24:
I pushed some stashed code I had from a while back here: https://github.com/gopcua/opcua/commit/c4ba7f2b22cacc61f9dbd69e5e3fea66963bef02

I _think_ it's ok. I had to tweak the extension struct gen code to remove a filter for types without any fields (e.g. `ua.PublishedDataSetSourceDataType`). Not sure why the filter was there in the first place.

kung-foo, 2020-07-24:
ref: https://github.com/gopcua/opcua/compare/fix/issue-277

magiconair, 2020-07-24:
Thanks. I've cleaned up the template to generate nicer empty structs and also update the `stringer` tool so that we don't run on different versions.

PR #358 (2020-07-24): Register all known extension objects
Fixes #277
closed 2020-07-24 by kung-foo