gopcua/opcua #281: StatusBadDecodingError (0x80070000) when browsing

kung-foo, 2019-09-26:
Server: Prosys OPC Simulation Server v3.2.0-214

Error message: `Decoding halted because of invalid data in the stream. StatusBadDecodingError (0x80070000)`

I see this when calling:
`attrs, err := node.Attributes(ua.AttributeIDNodeClass, ua.AttributeIDBrowseName, ua.AttributeIDDescription)`

(full source is loosely based on the new browsing example).

Note: this code _does_ (mostly) work against the python asyncio opcua server.

When looking at the packet trace, I see an error (frame 30, client to server) that shows a "malformed packet". Not sure if that is related.

[browse-error.pcap.gz](https://github.com/gopcua/opcua/files/3656597/browse-error.pcap.gz)

kung-foo, 2019-09-26:
FYI I tried applying https://github.com/gopcua/opcua/pull/282 but it had no effect.

kung-foo, 2019-09-26:
OK, I'm pretty sure this is the same issue as referenced here: https://github.com/gopcua/opcua/pull/200#discussion_r287770460

magiconair, 2019-09-26:
I think we can cache the namespace urls during connect and only refresh if we find something which we don't know about. This was going to bite us at some point might as well be now.

nbfhscl, 2020-02-20:
Is it on the agenda to solve this problem? I tried and succeeded in browsing ProsysServer after applying @kung-foo 's patch from #200.
In addition, browsing opcua foundation's DataAccessServer with the browsing example always returns a single node.

magiconair, 2020-02-20:
Generally, we want to be correct and provide a stable client library. So yes. 

However, I leave that up to @kung-foo at this point since I'm a bit swamped and can probably spend more time on this project in a couple of weeks. In the meantime, I'll try to fix small issues mostly focussing on the things affecting my project here.

magiconair, 2020-02-20:
We also accept PRs.

kung-foo, 2020-02-20:
Yeah, haven't had enough time to look any more at this. You can see my work around (without any patches) here: https://gist.github.com/kung-foo/9db8b4f942e5fa6d8d9902acbd6ac998#file-walk-go-L175-L186

closed 2025-04-28 by github-actions[bot]