gopcua/opcua #283: Prosys returns a DataValue with variant type == 0

kung-foo, 2019-09-26:
Server: Prosys OPC Simulation Server v3.2.0-214

When querying for an attribute (i.e. `ua.AttributeIDDescription`) Prosys may return a `DataValue` with the `HasValue` flag set _and_ with an embedded variant that has a type id of `0x00` (which is not listed in the spec). Looking at the same node in the Prosys UI shows Description=null.

This shows up as an error like: `invalid type id: 0`

Quick'n'dirty-non-standard-hack:

```diff
diff --git a/ua/enums.go b/ua/enums.go
index 6131fd0..73050d2 100644
--- a/ua/enums.go
+++ b/ua/enums.go
@@ -58,6 +58,7 @@ const (
 type TypeID byte
 
 const (
+       TypeIDNull            TypeID = 0
        TypeIDBoolean         TypeID = 1
        TypeIDSByte           TypeID = 2
        TypeIDByte            TypeID = 3
diff --git a/ua/variant.go b/ua/variant.go
index 3d0a32c..dc107c8 100644
--- a/ua/variant.go
+++ b/ua/variant.go
@@ -542,6 +542,7 @@ func (m *Variant) Time() time.Time {
 
 var variantTypeToTypeID = map[reflect.Type]TypeID{}
 var variantTypeIDToType = map[TypeID]reflect.Type{
+       TypeIDNull:            reflect.TypeOf(nil),
        TypeIDBoolean:         reflect.TypeOf(false),
        TypeIDSByte:           reflect.TypeOf(int8(0)),
        TypeIDByte:            reflect.TypeOf(uint8(0)),
```

kung-foo, 2019-09-26:
related prosys issue: https://forum.prosysopc.com/forum/opc-ua-simulation-server/simulator-returns-an-invalid-variant-type-id-for-some-attributes/

magiconair, 2019-09-26:
LGTM

—
Frank Schröder

PR #284 (2019-09-27): Add type id 0x00 support
fixes #283
closed 2019-09-27 by magiconair