gopcua/opcua #300: Variant.String() panics if underlying type is []string

kung-foo, 2019-11-08:
Using the convenience function `String()` when the underlying go type is not a literal `string` will result in a panic.

```go
func TestVariant(t *testing.T) {
	v := ua.MustVariant([]string{"a", "b", "c"})
	println(v.String())
}
```

```
--- FAIL: TestVariant (0.00s)
panic: interface conversion: interface {} is []string, not string [recovered]
	panic: interface conversion: interface {} is []string, not string
```

This is because we only check the opc type before casting and do not check array length or parameter.

At minimum we should change the convenience functions to return `($type, error)`. But I also think we should revisit the whole `Variant` interface since I am finding it a bit awkward to work with.

kung-foo, 2019-11-08:
I've been toying around with the idea of separating the value itself such that we can isolate that interface from the variant's need for supporting multidimensional arrays.

In my specific use case, I am converting the opc variant type into a protocol buffer. But protobufs don't support multidimensional arrays (which is a good thing IMO). So I need to encode/transport then as a flat array anyway.

```golang
package ua

type Value interface {
	Type() TypeID

	AsString() (string, error)
	AsBool() (bool, error)
	AsInt() (int, error)
	AsRaw() (interface{}, error)
	...
}

type Variant struct {
	Value // should Variant implement Value as a convenience for non-array nodes?

	typeID TypeID
	values []Value // maps directly to the layout of the encoded values

	...
}

func (v *Variant) Value() (Value, error) {
	return v.ValueAt(0)
}

func (v *Variant) ValueAt(dimensions ...int) (Value, error) {
	return nil, nil // do math to figure out element based on dynamic dimensions
}

func (v *Variant) Slice() ([]Value, error) {
	return v.SliceAt(0)
}

// this would only work for a slice of values. it would fail on a slice of slices. i think.
func (v *Variant) SliceAt(dimensions ...int) ([]Value, error) {
	return nil, nil // do math to figure out slice based on dynamic dimensions
}
```

magiconair, 2019-11-08:
Interesting idea. Lets fix the panic first and explore this a bit more.

PR #301 (2019-11-11): do not panic on arrays dimensions greater than zero
fixes #300 
note: this is a short term fix until a better interface can be designed.
closed 2019-11-11 by magiconair