gopcua/opcua #305: how to resolve it?

devicemate, 2020-01-03:
Error Info: The user identity token is valid but the server has rejected it. StatusBadIdentityTokenRejected (0x80210000)

devicemate, 2020-01-03:
I have set the opc ua server trusted

dwhutchison, 2020-01-03:
That sounds like your authorization profile isn't matching up with the server's.  

Can you tell me which authorization method you're using (eg. Anonymous, Username)?

Can you also show the output of the crypto example on your server using the `-list` flag?  It should tell you the valid combinations on all endpoints the server supports.

devicemate, 2020-01-06:
think you!  the authorization method is Certificate. 
- list
2020/01/06 09:03:56          Basic128Rsa15 | SignAndEncrypt  | (UserName,Anonymous,Certificate)
2020/01/06 09:03:56          Basic128Rsa15 | Sign            | (UserName,Anonymous,Certificate)
2020/01/06 09:03:56               Basic256 | Sign            | (UserName,Anonymous,Certificate)
2020/01/06 09:03:56                   None | None            | (UserName,Anonymous,Certificate)
2020/01/06 09:03:56               Basic256 | SignAndEncrypt  | (UserName,Anonymous,Certificate)
2020/01/06 09:03:56         Basic256Sha256 | Sign            | (UserName,Anonymous,Certificate)
2020/01/06 09:03:56         Basic256Sha256 | SignAndEncrypt  | (UserName,Anonymous,Certificate)

sec-policy is Basic256,
sec-mode is SignAndEncrypt
auth-mode is Certificate
the authorization profile is auto generate by  first , then I  have set the opc ua server trusted.
The second time I use the generated authorization file to connect to OPC UA server
Errors still occur

dwhutchison, 2020-01-06:
Thanks for the info.  The server looks like it should support what you're trying to do.

Can you tell me which server you're using?

With Certificate auth, you typically need to also trust the user certificate as well as the policy certificate.  You'll also need to use the Config option `opcua.AuthCertificate(cert)` to load the auth certificate separately from the policy certificate.  Even if the same certificate is used for both purposes, the server will typically still require you to trust both (once as a client, the second time as a user).

If you've trusted both certificates and it's still not working, can you share your connection code and a wireshark dump of the connection attempt (I realize a lot of it will be encrypted but it still helps)?  Find me on keybase (user: dwhutchison) if you don't want to post the files publicly.

magiconair, 2020-07-24:
I'll close this because of age. Please feel free to re-open if necessary.

closed 2020-07-24 by magiconair

boundray, 2022-08-05:
> think you! the authorization method is Certificate.
>
> * list
>   2020/01/06 09:03:56          Basic128Rsa15 | SignAndEncrypt  | (UserName,Anonymous,Certificate)
>   2020/01/06 09:03:56 [...]

The same operation, encountered the same problem！
Take the liberty to ask, how did you solve it in the end？