gopcua/opcua #310: Feature Request - examples for all examples.

fdamador, 2020-01-11:
Not an issue, but I'm wanted to have more examples of the examples.   I'm able to apply most of the examples but some are coming back with errors and I think it's just way I'm applying the code.

For instance, I'm able to read a tag value just fine.

Command: `C:\GoProjects\src\github.com\gopcua\opcua> go run examples/read/read.go -endpoint opc.tcp://localhost:49380 -node 'ns=2;s=root.PS1.DB101:REAL:66'`
Answer: `3`

But I'm stuck when wanting to write to this tag.  I appears that we need the tag identifier ID number not the actual tag name.

Command: `C:\GoProjects\src\github.com\gopcua\opcua> go run examples/write/write.go -endpoint opc.tcp://localhost:49380 -node 'ns=2;s=root.PS1.DB101:REAL:66' -value '4.2'`
Answer: `The value supplied for the attribute is not of the same type as the attribute's value. StatusBadTypeMismatch (0x80740000)`

As a result, can someone add example codes for all the example codes?

magiconair, 2020-01-11:
The examples are code examples not apps which are ready to use. I would guess that the value you are trying to write is a float value but the code in `examples/write` will try to write it as a string. You have to change the code to make this work with a float value. E.g. use `strconv.Atof` to convert and then create the variant from the float.

See https://github.com/gopcua/opcua/blob/master/examples/write/write.go#L40-L43

fdamador, 2020-01-12:
Understood, so the string type conversion has to match the value type for the opc ua tag.   I eventually, used the following with good result.

`input, _ := strconv.ParseFloat(*value, 32)`
`result := float32(input)`
`v, err := ua.NewVariant(result)`

Then it worked.

closed 2020-02-11 by magiconair

fdamador, 2020-03-27:
Thank you for uploading all the code examples a month ago.

I know this issue is closed but I was wondering why half of the server example code is commented it out. Is it because it's a work in progress code?

magiconair, 2020-03-28:
The server code is on a branch developed by @dwhutchinson but has been stale for a while. Right now we don’t have a working OPC/UA server.

fdamador, 2020-03-28:
Understood,  I was just trying to integrate this to the influxdata platform.  I’ve already submitted a simple [OPC ua client](https://github.com/influxdata/telegraf/pull/7213)