gopcua/opcua #311: OPCUA server is disconnecting regulary

juhiajmera13, 2020-01-21:
Hi 

I am trying to connect to a opc-ua server using python-opcua. The problem is, the client is getting shut down every now and then without communicating. How can I resolve this?

magiconair, 2020-02-11:
Hard to tell without more info. What I have observed is that when I have a subscription without any update then the subscription times out and the client disconnects. My workaround is to subscribe to a node that always changes, e.g. the server time. On a Siemens PLC this is `i=2258`.

Kowiste, 2020-03-19:
This is the opc for Go, I have your same problem and fix it using this option when create the client:

`c := opcua.NewClient(EndPoint, opcua.SecurityMode(ua.MessageSecurityModeNone), opcua.SessionTimeout(0))`
	`err := c.Connect(ctx))`

Setting the SessionTimeout to zero.

alexlev2004, 2020-11-04:
Check out #396 - it will disconnect regularly as there is a problem when SecurityMode=MessageSecurityModeNone.
I have a fix in a fork + open PR

closed 2025-04-28 by github-actions[bot]