gopcua/opcua #315: no matching versions for query "latest"

darlandieterich, 2020-01-31:
Why dont work for me when run example?

`go run examples/datetime/datetime.go -endpoint opc.tcp://localhost:4840`

ERROR:
> no matching versions for query "latest"

magiconair, 2020-02-01:
What server are you testing with?

—
Frank Schröder

darlandieterich, 2020-02-03:
Hi @magiconair, well, I'm beginner with opcua. I'm testing with `go run examples/server/server.go -endpoint opc.tcp://localhost:4840` 
but the message fail show: 

> cannot load github.com/gopcua/opcua/id: no matching versions for query "latest"

darlandieterich, 2020-02-03:
Sorry, I'm wrong,  I'm using go v1.11 and for this case necessary update my golang to v1.13, and generate again the script.

it would be interesting to add in the readme a minimum requirement for go version v1.13

closed 2020-02-03 by darlandieterich

magiconair, 2020-02-03:
Ah, the server code in gopcua is not implemented yet.

magiconair, 2020-02-03:
Will update the README.