gopcua/opcua #316: EOF exit status 1

darlandieterich, 2020-02-03:
Hi! 
My server running in `go run examples/server/server.go -endpoint opc.tcp://localhost:4840`

When connect my client in server `go run examples/datetime/datetime.go -endpoint opc.tcp://localhost:4840`

The client is down exiting with message: `EOF exit status 1`

magiconair, 2020-02-03:
The server code is not implemented yet.

darlandieterich, 2020-02-03:
You have one tiny functional example client and server?

magiconair, 2020-02-03:
We don't have a server. That is a left-over from the older code and should probably be removed until we have a functional server implementation. 

We have a functional client library which covers a good deal of the standard and is being used in production. One of the contributors is working on a server implementation on a branch but it is far from being finished. 

Please feel free to contribute a server implementation.

darlandieterich, 2020-02-03:
@magiconair very thanks for explanation. I will go contribute and share my experience with community. As soon as possible.

closed 2020-02-03 by darlandieterich